freifunk-gluon/gluon #899: USB Support

Byggvir, 2016-10-03:
gluon branch v2016.1.x

TP-Link wr1043v2 and wdr3600v1

Packages from default:
```
src/gz chaos_calmer_base http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base
...
```
I tried to install basic usb support. Installing kmod-usb-core works, but installing kmod-usb2 fails with the following error:

```

root@su-rhb-zingsheim-31-az:~# opkg install kmod-usb2
Installing kmod-usb2 (3.18.23-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages /base/kmod-usb2_3.18.23-1_ar71xx.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb2:
 *      kernel (= 3.18.23-1-b2f200610f46d20ef52d269421369d0c) * 
 * opkg_install_cmd: Cannot install package kmod-usb2.
```

rotanid, 2016-10-03:
because gluon compiles its own kernel, kernel packages have to be included or at least compiled directly with gluon itself. kernel modules have to match the kernel pretty close to be able to work.
you may include the modules in the file "site.mk", but you probably have to restrict inclusion to devices with enough flash space.
an extended site.mk that does this can be found in the repositories of many freifunk communities, for example here: https://github.com/tecff/site-ffa/blob/master/site.mk

NeoRaider, 2016-10-03:
Gluon's kernel is incompatible with the kernel modules from the OpenWrt repository. For that reason, a Gluon build will not only generate images, but also an opkg repo with all kernel modules; the Getting Started Guide describes what to do with this repository.

If you are using an image built by someone else, please ask the firmware builders to provide the repository (and optimally, even add it to site.conf so it will be work by default).

closed 2016-10-03 by NeoRaider

rubo77, 2016-10-04:
It is easy to add to the site.conf.

See http://git.freifunk.in-kiel.de/ffki-site/tree/?h=experimental