freifunk-gluon/gluon #903: mesh on lan does not add eth0 as mesh interface to

Byggvir, 2016-10-16:
I'm using mesh on lan / wan between fivel TP-LINK CPE 210 and TL-WR841v9 / v10 pairs

In 10th Oct I build a new image. Upgrading the node breaks the mesh_on_lan function.

I use following setup:

**CPE 210** 

With mesh on wan enabled:

root@su-rhb-test-05:~# batctl if
br-wan: active
mesh0: active
mesh-vpn: active
primary0: active

**WR841V9**
With mesh on lan enabled

root@su-rhb-test-02:~# batctl if
mesh0: active
mesh-vpn: active
primary0: active

Enabling eth0 on the WR841 with _batctl if add eth0_ manually starts the meshing. 

This looks like issue #875, but with my image from 03.09.2016 it works fine.

Byggvir, 2016-10-16:
I tested mesh on lan between a TP-LINK TL-WR841v10 and TL-WR842v2. 

I noticed the same problem. But on the WR842 you have to activate meshing on _eth1_  instead of _eth0_.

rubo77, 2016-10-16:
Are you sure, it's your build? And not the same reason as issue #875?

Was your cable plugged in while you first time started the routers after flashing?

Byggvir, 2016-10-16:
In my case it's not the cable.  I skip the setup page. My cables a plugged in all the time. I updated the routers from remote via scp /ssh. 

Under https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration I found the commands to setup _mesh on lan_ later. In 2 of 4 cases the _mesh on lan_ didn't work after the sysupgrade. But it worked after I executed the commands following

`
uci set network.client.ifname="bat0"
uci set network.mesh_lan.auto=1
uci commit network
/etc/init.d/network restart
`
Only in case five I assume a broken cable because it didn't work before either.

My question is why could got the wan configuration lost after a sysupgrade.

NeoRaider, 2016-10-16:
How was the mesh-on-LAN set up before the upgrade? Was `mesh_lan` enabled from the `site.conf` in the intial configuration, or was it later enabled using SSH?

Byggvir, 2016-10-28:
Mesh on LAN was not enabled in the sie config. I habe implemented a new Image and everything wenn welle. I propose to close this issue.

closed 2016-10-28 by Byggvir