freifunk-gluon/gluon #911: Ubiquiti Unifi AC Lite: Problems with eth0 persist

smoe, 2016-10-21:
We found this issue with the AC Lite already mentioned on https://github.com/freifunk-gluon/gluon/issues/689 : the wlan works fine, but the ethernet connection fails. A tcpdump indicated no traffic on the connection at all. In contrast, the latest Lede-image is running flawlessly.
Could that driver of Lede possibly be backported?

Best regards,

Zafer and Steffen

rotanid, 2016-10-23:
what is the "0.6.2" release?
we need to know which exact gluon version you or your community are using.

smoe, 2016-10-23:
I checked out tag v2016.2 and the git log's most recent entry says
```
commit 6964d30ff24eef9fc436e97bc909dc4c5e048ad4
Author: Matthias Schiffer <mschiffer@universe-factory.net>
Date:   Wed Sep 21 20:04:57 2016 +0200

    docs, README: change current release to v2016.2
```
Many thanks for looking into this!

NeoRaider, 2016-11-04:
Thanks for the report, please provide the contents of:
* /etc/config/network
* /tmp/sysinfo/board_name
* /tmp/sysinfo/model

Garunda, 2016-11-24:
Hello,

here are the desired file-contents:
/tmp/sysinfo/board_name:
`unifiac-lite`
/tmp/sysinfo/model:
`Ubiquiti UniFi-AC-LITE`

The file /etc/config/network content is attached.
[network.txt](https://github.com/freifunk-gluon/gluon/files/612232/network.txt)

SunboX, 2016-11-27:
Same for me! 😢  Please find the mentioned files attached.
[model.txt](https://github.com/freifunk-gluon/gluon/files/615043/model.txt)
[board_name.txt](https://github.com/freifunk-gluon/gluon/files/615042/board_name.txt)
[network.txt](https://github.com/freifunk-gluon/gluon/files/615044/network.txt)

ecsv, 2016-11-27:
I had some problems in the past with the AR803x PHYs. There are patches in LEDE which work around them. For example  https://git.lede-project.org/?p=source.git;a=blob;f=target/linux/generic/patches-3.18/735-net-phy-at803x-fix-at8033-sgmii-mode.patch;h=4a8f532aa36f729339e01ebadd6ea3cb292d7ac9;hb=HEAD

There are also other RFCs which I've submitted in the past related to SGMII with this PHY (but with QCA955x as SoC) like https://patchwork.ozlabs.org/patch/606468/ (and other changes in this series). But these aren't in LEDE and thus I would expect to LEDE to be still affected.

SunboX, 2016-11-28:
The fix was apllied to my router by @ecsv (http://vogtland.freifunk.net/map/#!v:m;n:802aa848b87c)

Uplink seems to be stable by now on the Config-Mode works also using DHCP.

closed 2016-11-29 by NeoRaider

smoe, 2016-12-29:
Heya @NeoRaider,
These changes appear not made it to the 0.6.2.2 release tag while @Garunda's  experimeal build just works smoothly. Since it is not so straight forward to reflash the device without Ethernet access, please kindly consider to forward those Ubiquity AC Lite changes to the next release.
Cheers,
Steffen

NeoRaider, 2016-12-30:
@smoe, the changes are in v2016.2.x; make sure you always `make update` and `make clean` after pulling.

PR #949 (2016-11-29): kernel: add at803x workaround for sgmii mode
Some (possibly broken) bootloaders (e.g. Ubiquiti UAP-AC-LITE) incorrectly initialize the at8033 phy. This breaks auto-negotation on these SGMII devices. The mode has therefore to be set back in to a valid configuration by Linux.

Fixes #911

----

This is just a possible fix for the problem from #911. I am still waiting for confirmation. But at least I could see a working uplink on http://vogtland.freifunk.net/map/#!v:m;n:802aa848b87c
commit 3d9a3f35 (2016-11-29): kernel: add at803x workaround for sgmii mode (#949)

Some (possibly broken) bootloaders incorrectly initialize the at8033 phy.
This breaks auto-negotation on these SGMII devices. The mode has therefore
to be set back in to a valid configuration by Linux.

Fixes #911
PR #950 (2016-11-29): v2016.2.x: kernel: add at803x workaround for sgmii mode
Some (possibly broken) bootloaders (e.g. Ubiquiti UAP-AC-LITE) incorrectly initialize the at8033 phy. This breaks auto-negotation on these SGMII devices. The mode has therefore to be set back in to a valid configuration by Linux.

Fixes #911

----

This is just a possible fix for the problem from #911. I am still waiting for confirmation. But at least I could see a working uplink on http://vogtland.freifunk.net/map/#!v:m;n:802aa848b87c
commit d54c9c5f (2016-11-29): kernel: add at803x workaround for sgmii mode (#950)

Some (possibly broken) bootloaders incorrectly initialize the at8033 phy.
This breaks auto-negotation on these SGMII devices. The mode has therefore
to be set back in to a valid configuration by Linux.

Fixes #911