freifunk-gluon/gluon #914: status page requires referer

J0WI, 2016-10-25:
The graphs on a nodes status page requires a client to send it's full referer that graphs and live stats are shown.
It's generally a bad idea to use the referer for privacy reasons on client side and security reasons on server side (lack of integrity). Is there a way to get rid of this requirement?

jplitza, 2016-10-25:
Could you please elaborate on how exactly the statuspage depends on the Referer? I set `network.http.sendRefererHeader` to 0 in my Firefox and the statuspage still seems to work as before.

J0WI, 2016-10-25:
I have `network.http.referer.XOriginPolicy = 1` which causes the issue.
I can confirm that it works when referer is completely off (`network.http.sendRefererHeader = 0`). That is quite strange.

jplitza, 2016-10-25:
Still can't reproduce the issue. Can you verify that the problem really is a Referer header (not) being sent (for example using the Developer Tools accessible via F12)? What requests fail with what reason in which constellation of settings?

J0WI, 2016-10-25:
I have 10 request without this setting and only 4 with this setting on a clean Firefox Nightly profile.

H4ndl3, 2018-02-14:
I encountered the exact same problem. I too set `network.http.referer.XOriginPolicy` to `1` with the effect that the live stats aren't shown. When I set it back to `0` everything works as expected.

rotanid, 2018-02-15:
#1194

commit 88906f23 (2018-02-26): gluon-status-page: reimplement based on gluon-web

This new status page is significantly smaller than the old one. It always
loads its resources from the same host as the page itself, not requiring
cross-origin requests anymore.

It also uses the common i18n infrastructure of gluon-web.

Fixes #914
closed 2018-02-26 by NeoRaider