freifunk-gluon/gluon #917: Hint to used git commit

rubo77, 2016-10-31:
In config mode, you can see the exact installed gluon version with a version coding that is not obvious.

There should be a hint, which git commit was used to create the firmware

NeoRaider, 2016-11-01:
I think the encoding is obvious enough even if you don't know about `git describe`. Also, it's a very common format by now, as most software is using git for version control.

closed 2016-11-01 by NeoRaider

rubo77, 2016-11-11:
OK, nur für andere, die danach suchen hier zur info:

mit `git describe` bekomme ich z.B. 

    v2016.2.1-1-gc47514c

dann kann man reverse suchen mit

    git describe v2016.2.1-1-gc47514c --exact-match 

>fatal: no tag exactly matches 'c47514c53504877cff9d2cdae898aeeec562b6b9'

ist zwar nicht fatal, aber da hat man die commit id: `c47514c53504877cff9d2cdae898aeeec562b6b9`

NeoRaider, 2016-11-11:
`c47514c` (der Teil hinter dem `g`) reicht doch, die lange Commit-ID braucht man eher selten.

rubo77, 2016-11-12:
This is really strange, I built a firmware on base of my own fork of gluon: https://github.com/rubo77/gluon/commit/0f9a1a9  which is a fork of the 2016.2.1 tag with some small changes

Now the firmware I built works fine, but it shows a strange version:

    	2016.2~exp1611121145 / gluon-v2015.1-686-g0f9a1a9

Why does it still show 2015.1? I thought, maybe it takes the last release it can find in https://github.com/rubo77/gluon/releases so I added a new release there, but that didn't change anything. stil this strange version

NeoRaider, 2016-11-12:
It is using the tags in your local git repository; if you used `git fetch` to fetch a branch, also run `git fetch --tags` on the same remote to get the tags.

rubo77, 2016-11-13:
I already fetched all tags. they are there:

```
git tag --list
...
v2015.1.1
v2015.1.2
v2016.2.1-ffki
```

rotanid, 2016-11-13:
how and where do you build it? perhaps you would also need to include "--follow-tags" when doing "git push"

rubo77, 2016-11-13:
I build on a server where i did this:

    git clone https://github.com/freifunk-gluon/gluon
    Git  remote add rubo77 https://github.com/rubo77/gluon/
    Git Checkout rubo77/v2016.2.1-ffki
    Git pull
    
Then Checkout our site-ffki and Use the script in it to build

rubo77, 2016-11-13:
The problem was not, that I didn't use annotated tags. It is fine, if the last release is taken as the last annotated tag, so the problem in my repo was (like i've seen iin many others) that the tags that were released in the main gluon repo were not pushed to the local repository with

    git push --follow-tags

now everything porks fine in my gluon build server.

before:

    $ git describe --always
    v2015.1-686-g0f9a1a9

And now:

```
$ git pull
remote: Counting objects: 8, done.
remote: Total 8 (delta 0), reused 0 (delta 0), pack-reused 8
Unpacking objects: 100% (8/8), done.
From https://github.com/rubo77/gluon
 * [new tag]         v2016.1    -> v2016.1
 * [new tag]         v2016.1.1  -> v2016.1.1
 * [new tag]         v2016.1.2  -> v2016.1.2
 * [new tag]         v2016.1.3  -> v2016.1.3
 * [new tag]         v2016.1.4  -> v2016.1.4
 * [new tag]         v2016.1.5  -> v2016.1.5
 * [new tag]         v2016.2    -> v2016.2
 * [new tag]         v2016.2.1  -> v2016.2.1
Already up-to-date.

$ git describe --always
v2016.2.1-3-g0f9a1a9
```

This should be documented somewhere, that you need to push with `git push --follow-tags`

rotanid, 2016-11-13:
i wrote that tonight ;-)
but that has nothing to do with gluon, it's simply git usage.