freifunk-gluon/gluon #919: Build of master and 2016.2.x fails

A-Kasper, 2016-11-04:
The build failed for me, because of a wrong openwrt link... 

```
--2016-11-04 08:53:56--  http://mirror2.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.

Download failed.
--2016-11-04 08:53:57--  http://downloads.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.
```

seems like a wrong link....

NeoRaider, 2016-11-04:
This report is missing the actual failed download; mirror2.openwrt.org and downloads.openwrt.org are always tried as fallback when the real download was unsuccessful, it is normal that these fail.

A-Kasper, 2016-11-04:
`
--2016-11-04 08:53:56--  http://mirror2.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.

Download failed.
--2016-11-04 08:53:57--  http://downloads.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'linux-firmware-52442afee9907bc32a058f22bb3295d040677c26'...
fatal: unable to access 'https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/': gnutls_handshake() failed: Handshake failed
make[4]: *** [/home/kasper/gluon/build/ar71xx-generic/openwrt/dl/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz] Error 128
make[4]: Leaving directory `/home/kasper/gluon/openwrt/package/firmware/linux-firmware'
make[3]: *** [package/firmware/linux-firmware/compile] Error 2
make[3]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make[2]: *** [/home/kasper/gluon/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.package_compile] Error 2
make[2]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make[1]: *** [prepare] Error 2
make[1]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make: *** [all] Error 2
`

A-Kasper, 2016-11-04:
```

--2016-11-04 08:53:56--  http://mirror2.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.

Download failed.
--2016-11-04 08:53:57--  http://downloads.openwrt.org/sources/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-04 08:53:57 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'linux-firmware-52442afee9907bc32a058f22bb3295d040677c26'...
fatal: unable to access 'https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/': gnutls_handshake() failed: Handshake failed
make[4]: *** [/home/kasper/gluon/build/ar71xx-generic/openwrt/dl/linux-firmware-2016-01-25-52442afee9907bc32a058f22bb3295d040677c26.tar.xz] Error 128
make[4]: Leaving directory `/home/kasper/gluon/openwrt/package/firmware/linux-firmware'
make[3]: *** [package/firmware/linux-firmware/compile] Error 2
make[3]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make[2]: *** [/home/kasper/gluon/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.package_compile] Error 2
make[2]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make[1]: *** [prepare] Error 2
make[1]: Leaving directory `/home/kasper/gluon/build/ar71xx-generic/openwrt'
make: *** [all] Error 2

```

NeoRaider, 2016-11-04:
Hmm, I can access https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git just fine, you could try using `gnutls-cli` to find out why your connection fails.

A-Kasper, 2016-11-04:
`Resolving 'git.kernel.org'...
Connecting to '199.204.44.194:443'...
|<2>| ASSERT: gnutls_constate.c:695
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_AES_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_AES_128_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_CAMELLIA_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_AES_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_AES_256_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_CAMELLIA_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_RSA_3DES_EDE_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_AES_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_AES_128_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_CAMELLIA_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_AES_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_AES_256_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_CAMELLIA_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_3DES_EDE_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: DHE_DSS_ARCFOUR_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_AES_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_AES_128_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_CAMELLIA_128_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_AES_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_AES_256_CBC_SHA256
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_CAMELLIA_256_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_3DES_EDE_CBC_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_ARCFOUR_SHA1
|<3>| HSK[0x2361ad0]: Keeping ciphersuite: RSA_ARCFOUR_MD5
|<2>| EXT[0x2361ad0]: Sending extension SERVER NAME (19 bytes)
|<2>| EXT[0x2361ad0]: Sending extension SAFE RENEGOTIATION (1 bytes)
|<2>| EXT[0x2361ad0]: Sending extension SESSION TICKET (0 bytes)
|<2>| EXT[SIGA]: sent signature algo (4.2) DSA-SHA256
|<2>| EXT[SIGA]: sent signature algo (4.1) RSA-SHA256
|<2>| EXT[SIGA]: sent signature algo (2.1) RSA-SHA1
|<2>| EXT[SIGA]: sent signature algo (2.2) DSA-SHA1
|<2>| EXT[0x2361ad0]: Sending extension SIGNATURE ALGORITHMS (10 bytes)
|<3>| HSK[0x2361ad0]: CLIENT HELLO was sent [139 bytes]
|<2>| ASSERT: gnutls_record.c:726
|<2>| ASSERT: gnutls_record.c:1122
|<2>| ASSERT: gnutls_handshake.c:2762
*** Fatal error: A TLS fatal alert has been received.
*** Received alert [40]: Handshake failed
*** Handshake has failed
GnuTLS error: A TLS fatal alert has been received.
`

maybe it's related to 
http://stackoverflow.com/questions/13524242/error-gnutls-handshake-failed-git-repository
http://askubuntu.com/questions/186847/error-gnutls-handshake-failed-when-connecting-to-https-servers
https://answers.launchpad.net/ubuntu/+source/git/+question/255855

A-Kasper, 2016-11-04:
> It seems the gnutls version is too old for kernel.org, I can reproduce
> the issue on Debian Wheezy. :|
>
> The best way around this issue would be to use a newer version of
> libgnutls. Unfortunately, git uses libcurl3-gnutls, which is not availabe
> in a newer version in wheezy-backports, even though libgnutls itself can be
> updated from wheezy-backports. libcurl would need to be rebuilt against the
> new libgnutls version...
>
> This is unfortunate for Gluon 2016.2, which was intended to be buildable
> on Wheezy; for the master, it doesn't matter, we'll need a newer GCC
> version for the updated NodeJS host build anyways as soon as we switch to
> LEDE.

we could patch the url to http instead of https

xf-, 2016-11-08:
Use git:// - ssh works (if possible)
https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/
List of clone URLs git:// also https form google works. SSH seems like nicer way to go.

xf-, 2016-11-09:
This part of patch breaks wheezy
https://github.com/freifunk-gluon/gluon/blob/master/patches/openwrt/0007-mac80211-hostapd-iw-.-update.patch#L273

A-Kasper, 2016-11-09:
Am 09.11.2016 um 23:39 schrieb Xaver Maierhofer:
> This part of patch breaks wheezy
> https://github.com/freifunk-gluon/gluon/blob/master/patches/openwrt/0007-mac80211-hostapd-iw-.-update.patch#L273

good work! we just need to change https to http in this patch and
everything should work fine....

NeoRaider, 2016-11-10:
Thanks, I'll change it back from https to git.

closed 2016-11-10 by NeoRaider

thomaszbz, 2017-06-22:
That sounds pretty MITM-insecure.

rotanid, 2017-06-22:
as we don't support the GCC version of wheezy anymore, we cloud also change again to https (if we didn't already with the switch to LEDE)

NeoRaider, 2017-06-22:
@thomaszbz A specific SHA1 hash is checked out from the Git repository; the integrity checks of the Git protocol will prevent MITM.

Anyways, @rotanid is right, and we're back to https since switching to LEDE.

thomaszbz, 2017-06-22:
@NeoRaider Thats fine. I go along with your consideration as long as the SHA1 cannot be manipulated by a MITM.

Well, should still not stay like that in the long run: https://shattered.io/

However, TLS not necessarily does a better job here.

thomaszbz, 2017-06-22:
In patch d472604, that looks more like a MD5SUM:

[image]

MD5 should not be used to make something MITM-safe.

commit 60b4830c (2016-11-10): linux-firmware: switch back to git protocol from https for download

Using HTTPS breaks the download on Debian Wheezy. The switch was an
unintended side effect of the backport from LEDE.

Also, fix the commit message of the backport patch (mwlwifi is not updated
anymore since the last OpenWrt base upgrade).

Fixes #919
commit d472604a (2016-11-10): linux-firmware: switch back to git protocol from https for download

Using HTTPS breaks the download on Debian Wheezy. The switch was an
unintended side effect of the backport from LEDE.

Also, fix the commit message of the backport patch (mwlwifi is not updated
anymore since the last OpenWrt base upgrade).

Fixes #919