freifunk-gluon/gluon #925: Add Mesh on WAN/LAN status to Status Page

Nurtic-Vibe, 2016-11-09:
For debugging purposes it would be nice to have the Status of Mesh-on-WAN and Mesh-on-LAN listed on the Status Page

A-Kasper, 2016-11-09:
Am 09.11.2016 um 14:57 schrieb Hermann Lienstromberg:
> For debugging purposes it would be nice to have the Status of
> Mesh-on-WAN and Mesh-on-LAN listed on the Status Page

+1

LAN and WAN config and status should be shown:
- mesh activated?
- media status (connected or not)
- static ip or dhcp?
- internet reachable via mesh? ipv4 ipv6?
- or if not activated internet reachable via gateway ipv4 ipv6?

and next to the tunnelstatus the status of ipv4 or ipv6 gateway is
present.....

(this should not show internal IPs)

rotanid, 2016-11-20:
also, showing configured bandwidth limits would be nice

mokkin, 2017-01-02:
and PoE status too

rubo77, 2017-01-05:
After analysing the status-page code, I think this has to be implemented in the package `package/gluon-status-page-api/` in new files in the folder `files/lib/gluon/status-page/www/cgi-bin/dyn/`

There we could add scripts that output the needed data, so we can read them in the package `package/gluon-status-page/` in the file [`src/js/lib/streams.js` from line 46 on](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-status-page/src/js/lib/streams.js#L46).

rotanid, 2018-07-12:
let's track this in #644

closed 2018-07-12 by rotanid