freifunk-gluon/gluon #929: add explaining comments to lua source files

rubo77, 2016-11-10:
Now that all lua files are minified and comments are extracted from the source can someone please add comments to explain the scripts?

especially this would be really handsome to have some comments explaining the single variables used there:

https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-core/luasrc/usr/lib/lua/luci/controller/gluon-config-mode/index.lua#L61

NeoRaider, 2016-11-10:
Tickets like this won't do any good. Either you submit a PR adding comments, or you don't.

closed 2016-11-10 by NeoRaider

rubo77, 2016-11-10:
I was asking for someone, that understands the code to do this. I cannot add comments, If I don't understand the purpose there