freifunk-gluon/gluon #931: shorter image-names so the tp-link interface accepts them

rubo77, 2016-11-10:
the TP-LINK interface doesn't accept filenames longer than 64 characters.

this is easily getting too long, if you create images with the version for example `2016.2.1~experimental1611101200`

So maybe we can automatically create symlinks in the output folder of the Firmware images with shorter names to the tp-link images that have too long names. Or instead of adding the extension `-sysupgrade` just add `-upgrd` or such.

 see:

https://forum.openwrt.org/viewtopic.php?pid=334978#p334978

NeoRaider, 2016-11-10:
The sysupgrade image is the wrong one anyways, and we don't have a -factory suffix for this reason?

rubo77, 2016-11-10:
OK, so this issue can be closed then?

closed 2016-11-10 by NeoRaider