freifunk-gluon/gluon #933: ibss0 hickup in v2016.2++

viisauksena, 2016-11-11:
dont know if this is related to other open bugs ( like https://github.com/freifunk-gluon/gluon/issues/889 https://github.com/freifunk-gluon/gluon/issues/753 or not likely https://github.com/freifunk-gluon/gluon/issues/605 )... but this looks straight forward

as reported here 
https://forum.freifunk.net/t/announce-gluon-v2016-2-1
there are some routers which loose (at least?) ibss0 neighbours - i had this some times also, no logread , no dmesg so far. 
but a "iwinfo phy0 scan" solve the hickup instantly.

there is a complex proposed fix in the freifunk forum, i use for myself a simple iwinfo phy0 scan every hour since this shouldnt do any harm , 
but both are no solutions - just dirty hacks

viisauksena, 2016-11-12:
dont know if related,
everytime (as node is working) i make **iwinfo phy0 scan** i got this in logread / dmesg
```Sat Nov 12 09:25:00 2016 kern.info kernel: [16308.580000] IPv6: ADDRCONF(NETDEV_UP): tmp.phy0: link is not ready
```

rotanid, 2016-11-13:
https://forum.freifunk.net/t/wifihaenger-in-gluon-2016-2-x-quickfix/13821

viisauksena, 2016-11-13:
@rotanid du wolltest konkretes feedback, 
beobachtet mit dem master build zu dem zeitpunkt wo der 2016.2.1 rauskam, inkl make clean und make update im vorfeld. benutzt wird ibss0 und bat-v14 - aufgefallen an 2 unterschiedlichen Tplink 841 - die autoupdate hinter sich haben.

symptom : wegbrechen der mesh-links - simply dead ... ifconfig mit normaler ausgabe, nichts im logread, nichts im dmesg, aufgefallen weil der mit uplink (!) in einem dichten Meshnetz ohne Mesh-Kontakte war. 

Lösung - live beobachtet über die status page (die war bei ibss0 leer) - "iwinfo phy0 scan" .. danach sofort neu gefundene meshpartner in der status page. und logisch im meshviewer nach der entsprechenden Verzögerung.
daher der Bugname "hickup".

rotanid, 2016-11-13:
as i wrote in #889 , these are duplicates of #605 - please continue ath9k-related discussions there, feel free to copy your information that you wrote here today to the other issue.

closed 2016-11-13 by rotanid