freifunk-gluon/gluon #939: switch IP of gluon-client-bridge and local-node

christf, 2016-11-19:
- [ ] decide whether this is a good idea and use the above package for both babel and gluon networks
OR
implement a gluon-client-bridge-babel package that does this just for babel networks.

lemoer, 2016-11-23:
Could you elaborate why you want to switch the IPs?

christf, 2016-11-23:
the babel branch should hopefully be able to run without macvlan reducing complexity. Switching the IPs allows to
- use the same package (gluon-client-bridge) for batman and babel networks
- allow for the local-node feature in a babel network while keeping the node properly available in the client net. If not switched, Babel will distribute bogus routes via the local-node IP inside the net.

PR #957 (2016-12-10): Merge gluon-next-node into gluon-client-bridge, switch roles of br-client and local-node interfaces
MAC and IP addresses are switched. This makes the gluon-client-bridge package more useful for different routing protocols that don't need a unique address on the client bridge.

As a side effect, gluon-radvd is now using the next-node address, which had been considered before, but was dismissed to avoid having gluon-radvd depend on gluon-next-node and gluon-mesh-batman-adv. This will be useful for announcing default routes via gluon-radvd.

One downside is that this introduces a minor dependency [... cut]
closed 2016-12-10 by NeoRaider