freifunk-gluon/gluon #940: simple-tc: commit and reboot is necessary to make it work

Tarnatos, 2016-11-20:
gluon: 2016.2.1

To get the bandwidth limitation using simple-tc working, a commit and a reboot is necessary. This should be improved to make it more flexible and to incresse the flash lifetime.

https://forum.freifunk.net/t/bandbreitenbegrenzung-simple-tc-ohne-funktion/13855/

rotanid, 2016-11-20:
please make sure to also try
/etc/init.d/network restart
after the changes, first without comitting.
this could help without a reboot.

NeoRaider, 2016-11-20:
The values are applied when the interface appears, so restarting fastd should work (`/etc/init.d/network restart` won't have an effect).

Tarnatos, 2016-11-20:
Unfortunately, restarting fastd has no effect. I've tried it.

NeoRaider, 2017-02-11:
I can't reproduce this issue. Restarting fastd reliably applies the new settings, even without committing them.

closed 2017-02-11 by NeoRaider