freifunk-gluon/gluon #941: gluon-mesh-batman-adv-core: respondd: doesn't detect VLAN-on-IBSS interfaces as wireless

NeoRaider, 2016-11-20:


mokkin, 2016-11-21:
Please conside the forum entry 111 and following:
https://forum.freifunk.net/t/hopglass-development-thread/10984/111

mokkin, 2016-12-01:
Half work done. I built a test version and one direction is showing correctly "Wifi". But the other direction is still showing "cable". Just look at the example as the is no physical wired connection to the mesh, only by wlan:
http://gadow.freifunk.net:8018/hopglass/#!v:g;n:60e327ee3370

NeoRaider, 2016-12-01:
Did you update both sides of these links?

rotanid, 2016-12-01:
he, @mokkin, obviously hasn't, if the shown version numbers are correct ;-)

mokkin, 2016-12-02:
yes, you are right :) stupid of me to forget the other side... Here you can see with two new devices that it's working fine:
http://gadow.freifunk.net:8018/hopglass/#!v:g;n:ec086b2bbee4

closed 2016-12-03 by NeoRaider

rubo77, 2017-04-17:
Can you add some description, why this was a problem, that had to be fixed?

what was the impact before this was fixed?

NeoRaider, 2017-04-18:
@rubo77 When VLAN-on-IBSS was used instead of plain IBSS, all wireless mesh links were announced as "other"/wired and displayed as such in maps. Very few sites use VLAN-on-IBSS.

PR #954 (2016-12-03): gluon-mesh-batman-adv-core: respondd: correctly detect lower interface types
Fixes #941

@mokkin: This is only compile tested, place check if this fixes your issue.
commit b32ed384 (2016-12-03): gluon-mesh-batman-adv-core: respondd: correctly detect lower interface types (#954)

Fixes #941
commit 185d3eb8 (2016-12-03): gluon-mesh-batman-adv-core: respondd: correctly detect lower interface types (#954)

Fixes #941