freifunk-gluon/gluon #945: status-page: preparing for babel

lemoer, 2016-11-25:
TODOs
----

**gluon-mesh-babel:**

- [ ] create a respondd module
- [ ] respondd-module: implement nodeinfo.network.mac
- [ ] respondd-module: implement nodeinfo.network.mesh
- [ ] respondd-module: implement nodeinfo.network.addresses
- [ ] create an issue to extend the respondd module for gluon-mesh-babel (see notices)

**status-page-api:**

- [x] add "protocol" field to /cgi-bin/dyn/neighbours-batadv
- [x] provide information about babel neighbours in /cgi-bin/dyn/routing-metrics
- [ ] rename to /cgi-bin/dyn/routing-metrics
- [ ] use ubus call to get network interfaces for mesh ifaces in /cgi-bin/interfaces
- [ ] use ubus call to get network interfaces for mesh ifaces in /cgi-bin/dyn/stations?iface
- [ ] use ubus call to get network interfaces for mesh ifaces in /cgi-bin/dyn/neighbours-nodeinfo?iface

**status-page:**

- [ ] rename /cgi-bin/dyn/neighbours-batadv to /cgi-bin/dyn/routing-metrics
- [x] differentiate between batman and babel metrics and show them differently
- [ ] test in babel
- [ ] test in batman-adv

**general:**

- [ ] docs: add documentation about gluon-status-page

Notices:
--------

- ubus call to get network interfaces:
    - ```ubus call network.interface dump | jsonfilter -e "@.interface[@.proto='gluon_mesh' && @.up=true].device"```

- these resources remain unimplemented in the gluon-mesh-babel respondd provider:
    - nodeinfo.software.babeld
    - statistics.clients
    - statistics.traffic
    - statistics.gateway
    - statistics.gateway_nexthop
    - neighbours.*

Questions:
----------

- Why do we have providers/neighbours-batadv instead of using respondd neighbours?

christf, 2016-11-25:
2. can be answered with: https://github.com/freifunk-ffm/gluon/blob/n_babel-work/package/gluon-mesh-babel/files/lib/gluon/upgrade/980-gluon-mesh-babel-firewall#L9

lemoer, 2016-11-25:
@rotanid Could you assign me on these issue?

lemoer, 2016-11-26:
The status-page depends on the respondd facts ```nodeinfo.network.mesh.*.interfaces[] = $mac``` to find out which wifi/mesh neighbour has ```$mac```:

```
{ "nodeinfo": { "network": { "mesh": {

    "bat0": {
        "interfaces": {
            "tunnel": [
                "66:6a:b3:de:dd:58"
            ],
            "wireless": [
                "66:6b:b5:de:dd:58",
                "66:6b:b4:de:dd:58"
            ],
            "other": [
                "66:67:b4:de:dd:58"
            ]
        }
    }

} } }
```

As ```bat0``` is the interface name, we have a problem in babel to hold the convention here. We do not have an equivalent to such an interface in babel. How should we name this section in babel?

An unbeautiful solution would be using ```nodeinfo.network.mesh.babel.interfaces[]```...

christf, 2016-11-26:
actually since in theory there could be more than one mesh protocol on one node, using  nodeinfo.network.mesh.babel.interfaces[] does not seem so ugly to me. this would also allow for leaving bat0 as it is even though it seems sensible to call it batman instead of bat0.

christf, 2016-11-26:
4. br-client is for the moment still existing in babel however there seems to be some pull to remove it. at a later stage.
5. this is probably possible but I do not think this is a good idea - what if more than one mesh protocol is used on one node in the future? Shouldn't each report its own metrics?

christf, 2016-12-08:
#745 is an early work which will likely be superseeded by the results of the work going into this issue.

lemoer, 2016-12-21:
Back to the question how we get the mesh ifaces:
- How do we solve this in shell?
- Isn't there a lua function which is exported by gluon_util?
    - If not, do we want to do this?

lemoer, 2016-12-21:
What is the current status of the ```br-client``` interface? (Q4)

christf, 2016-12-21:
br-client will remain in gluon for quite some time. The gluon-client-bridge-babel has been obsoleted by the change in gluon-client-bridge. This was taken from an irc discussion from last week.

I feel find_meshes() is the way to go in lua-context. However I do not know where this function should be placed. If you provide the place @NeoRaider, I will put it there as part of PR #934.

christf, 2016-12-21:
as discussed on irc with @neoraider he liked the ubus version that must be converted to lua and should be placed in gluon.util. I took the liberty of implementing this in PR #990

christf, 2017-01-07:
I have just pulled the rebase_statuspage branch into the babel-PR and will continue the integration there.

christf, 2018-11-26:
closing because nothing is left to do for this PR.

closed 2018-11-26 by christf