freifunk-gluon/gluon #948: No DHCP-WAN-IP with GBit on 8169SC (x86-generic)

Garunda, 2016-11-26:
I found an issue with the tag v2016.2 release. The device is a Fujitsu S550 with an PCI Ethernet Card (RTL 8169SC). As WAN-Interface is per default the PCI-NIC configured. We do not get an IP when using an GBit-switch attached to the WAN-interface. Everything works fine when we are using an 100mbit-switch. 

Best regards

Zafer

rotanid, 2016-11-28:
with latest v2016.2.x, it is working in our Futro S550 with a cheap Realtek PCI NIC attached to a gigabit switch.
haven't checked older builds, please also try the newest!
have you considered a faulty gbit-switch? try a second switch.

Garunda, 2016-11-28:
* checked with current master: same problem
* the problem occured with 2 different switches on 2 different locations
* connecting direct to an AVM Fritz!box 7390: No problem
* connecting direct to an TP-Link TL-WDR3600 with Freifunk Fw: No problem
* toggle eth1 and eth0 on the S550: No problem
* WR1043, WDR3600 or Bullet M2 have no problem at these switches
* I found no other devices which have problems at these switches, but the S550 with 8169SC
* tested 3 different S550 +8169SC
--> to it must be an problem with some switches and 8169SC
I have set up 3 locations with S550: 1 is running without problems, the other 2 have these problem. --> 33% chance to get is working at an new location --> not really nice

--> What can we do to find out what is going wrong. Which information do we need?

rotanid, 2016-11-29:
was there a gluon version before, that didn't have this issue? if so, which one specifically?
maybe you could try, if the problem still exists using the latest LEDE nightly build?
if the answer to one of those two questions is yes, there's a good chance to solve this.

do you happen to know the exact models of the two switches with which the problem occured?

Garunda, 2016-12-07:
The following devices cause the problem:
* Hitron CVE-30360 H/W-Version 1B
* Zyxel GS-108B
I have not tested the latest LEDE nightly build or older gluon releases yet, so be patient.

Garunda, 2016-12-28:
I have tested:
* Latest nightly LEDE build: doesn't show the issue.
* I didn't found an older gluon-release without the problem, but i haven't tested all.

NeoRaider, 2017-01-09:
Okay, I hope we can get the Gluon master switched to LEDE soon, so this problem will be solved as well.

mweinelt, 2017-02-07:
@Garunda Did you have a chance to test Gluon with the latest version of LEDE to verify this works now?

Garunda, 2017-03-12:
Hello,

i have tested the latest Gluon at all problematic hardware -> Works fine. The problem is solved

closed 2017-03-12 by rotanid