freifunk-gluon/gluon #952: mesh_lan interface not coming up after re-connecting cable (possible netifd bug)

rotanid, 2016-11-28:
situation:
image built based on the v2016.2.x branch
mesh-on-lan activated and running

bug:
if cable gets disconnected during operation, mesh-on-lan doesn't work after re-connecting the cable.
the gluon_mesh proto script is called, but the log shows no sign of batman_adv getting activated again on the interface.
ifstatus mesh_lan also shows the interface stays down.

workaround:
call 'ifup mesh_lan' manually

side-note:
if the cable isn't connected during boot, attaching it enables mesh-on-lan just fine for the first time, following disconnections trigger the bug.

this might be related to #635 and #905 and might be fixed by updating to a newer version of netifd

rotanid, 2016-11-30:
seems to happen only in conjunction with a backport of batman-adv 2016.4, low priority for now

ecsv, 2016-11-30:
Just tried to bring a OM5P in this state and then doing:

<pre>
export IFNAME=br-mesh_lan
export FIXED_MTU=1
export TRANSITIVE=1
/lib/gluon/mesh-batman-adv-core/config_mesh_interface setup 1532
</pre>

The script didn't return an error but also didn't add the interface to the bat0 interface. Now lets try:

<pre>
$ echo bat0 > /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
none
</pre>

And now lets try the fix from https://www.open-mesh.org/issues/307 (or basically all patches from the never-released batman-adv 2016.4.1). 

<pre>
# port plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
bat0
# port not plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
none
# port plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
bat0
# port not plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
none
# port plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
bat0
# port not plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
none
# port plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
bat0
# port not plugged in
$ cat /sys/class/net/br-mesh_lan/batman_adv/mesh_iface
none
....
</pre>

This is already fixed in #916 (see commit "batman-adv: add patches from 2016.4-maint 2016-10-29")

closed 2016-12-11 by NeoRaider