freifunk-gluon/gluon #956: Add support for Linksys WRT1200AC

LuiKast, 2016-11-30:
The Linksys [WRT1200AC](https://wiki.openwrt.org/toh/hwdata/linksys/linksys_wrt1200ac) is a quite interesting device because it provides excessive CPU power (Marvell Armada 1,33 Ghz Dual-Core), what qualifies it as the perfect offloader for larger installations but it lacks gluon support right now. 

It's a little bit more expensive (75 USD, 119 EUR) than used Futro S550 computers, but it could be even more powerful and is a normal router instead of a reconfigured personal computer. Benchmarks [show](https://www.privateinternetaccess.com/forum/discussion/comment/33308/#Comment_33308), that the WRT1200AC is able to handle more than 50 Mbit/s with OpenVPN. Converted to salsa2012+umac we might expect something between 100 and 200 Mbit/s (Gbit-WAN-Port).

Currently the WRT1200AC v1 is supported by [OpenWRT](https://wiki.openwrt.org/toh/linksys/wrt_ac_series). A few months ago it was replaced by a v2, which requires a [newer mwlwifi driver](https://dev.openwrt.org/changeset/49404) for wifi but the rest is still the same hardware.

belzebub40k, 2016-12-01:
Gluon support seems relatively easy. The currently used OpenWrt [sources](https://github.com/openwrt/openwrt/blob/0f757bd2606971252f901ef3faf4dbd0086315f7/target/linux/mvebu/image/Makefile#L156) already include support for the WRT1200AC v1. But the new mwlwifi 10.3.0.18-20160804 driver which is needed for v2 is not included by now. But as this is just [one line](https://github.com/openwrt/openwrt/blob/0f757bd2606971252f901ef3faf4dbd0086315f7/package/kernel/mwlwifi/Makefile#L20) to change this shouldn't be a problem. Do you know if the new driver version works properly with a v1 device or do you have hardware to test it on?

LuiKast, 2016-12-01:
I haven't bought a WRT1200AC yet, because there is only the v2 in stock and I don't want to buy it, when there is no support for it. If there will be support from gluon for this device, the team at Freifunk Stormarn will build a firmware. So if the gluon developers can and will add support for the WRT1200AC v2, I would buy one and if it helps, I could bring it to an appointment in the gluon city of Lübeck.

belzebub40k, 2016-12-02:
I ordered a warehouse deal from amazon yesterday a the device looks very interesting in general. We will see if it is v1 or v2. Both should work according to the OpenWrt sources used by Gluon.

belzebub40k, 2016-12-02:
Ok, images are building fine. Now I need to wait for my device which should be delivered tomorrow but I'm afraid I wont find time to test it this weekend.

belzebub40k, 2016-12-04:
According to my first tests Gluon runs fine on the WRT1200AC v2 and the fastd throughput in my first tests was really good with 40 Mbit/s. But there is one downside the mwlwifi drivers does not support meshing via IBSS or 802.11s.

LuiKast, 2016-12-04:
Are you sure, that there was no bottleneck while measuring (internet connection, gateway, VPN-exit)? Other users reported > 50 Mbit/s with OpenVPN with the WRT1200AC. Converted to salsa2012+umac we should expect more than 100 Mbit/s. Just 40 Mbit/s would be a huge disappointment.

belzebub40k, 2016-12-04:
As said this was just a first test. Will do a proper test when I've more time. But 40 Mbit/s is already better then every MIPS device that is already supported by gluon.

belzebub40k, 2016-12-05:
I opened two pull request to add the target for the WRT1200AC.

[libplatforminfo: add target mvebu](https://github.com/freifunk-gluon/packages/pull/151)
[Add support for mvebu target](https://github.com/freifunk-gluon/gluon/pull/964)

NeoRaider, 2016-12-05:
Note that batman-adv over VPN is always significantly worse than just the VPN itself (compare real-life bandwidth with batman-adv over fastd with the salsa2012+umac benchmarks on https://projects.universe-factory.net/projects/fastd/wiki/Benchmarks )

belzebub40k, 2016-12-05:
Looks like 802.11s support won't be implemented in the near future.

https://github.com/kaloz/mwlwifi/issues/127

belzebub40k, 2016-12-05:
I did some more tests on the fastd connection. I connected the WRT1200AC to my PC (1Gbit/s) and created a fastd tunnel with a MTU of 1406 and batman-adv on top between them. Running netperf from a Laptop that is connected to the WRT1200AC via cable shows an average throughput of 90 Mbit/s.

LuiKast, 2016-12-06:
Thank you @belzebub40k for your testing efforts and for adding the gluon-target. It is good to know, that the WRT1200AC does not perform better than a Futro S550. Thank you both for your work on that matter!

closed 2016-12-06 by NeoRaider