freifunk-gluon/gluon #959: babel-mesh requires igmp_snooping = 0 on br-client

christf, 2016-12-01:
This issue is to drive the analysis of the issue and track the progress on getting a babel mesh with l3roamd to work properly with igmp_snooping on br-client.

@T_X let us not run the analysis inside the PR, let us instead run this here.
@rotanid, would you please assign this issue to me?

christf, 2016-12-03:
This dump was taken with igmp_snooping=1 it took more than 30 seconds before the first traffic even did show up on the bridge. It took a long time before the client actually grabbed an IP address

`
./tcpdump  -p -i br-client
tcpdump: WARNING: br-client: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-client, link-type EN10MB (Ethernet), capture size 65535 bytes

[ 1542.240000] eth0: link up (1000Mbps/Full duplex)
[ 1542.240000] br-client: port 1(eth0) entered forwarding state
[ 1542.250000] br-client: port 1(eth0) entered forwarding state
[ 1544.250000] br-client: port 1(eth0) entered forwarding state
[ 1555.520000] REJECT(dest wan)IN= OUT=br-wan SRC=fe80:0000:0000:0000:6670:02ff:febb:db01 DST=ff02:0000:0000:0000:0000:0000:0000:0002 LEN=72 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=131 CODE=0 
[ 1557.020000] REJECT(dest wan)IN= OUT=br-wan SRC=fe80:0000:0000:0000:6670:02ff:febb:db01 DST=ff02:0000:0000:0000:0000:0001:ffbb:db01 LEN=72 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=131 CODE=0 
[ 1557.160000] REJECT(dest wan)IN= OUT=br-wan SRC=fe80:0000:0000:0000:6670:02ff:febb:db01 DST=ff02:0000:0000:0000:0000:0001:ff00:0000 LEN=72 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=131 CODE=0 

13:41:05.733038 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1: ICMP6, router advertisement, length 80
13:41:06.782382 IP6 fe80::9eeb:e8ff:fe20:92a2.42913 > nextnode.domain: 55057+ SOA? local. (23)
13:41:06.848918 IP6 nextnode.domain > fe80::9eeb:e8ff:fe20:92a2.42913: 55057 NXDomain 0/1/0 (98)
13:41:09.332006 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.44965 > ber01s14-in-x04.1e100.net.https: UDP, length 1350
13:41:09.332485 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59668 > ber01s14-in-x04.1e100.net.https: Flags [S], seq 1628829021, win 28800, options [mss 1440,sackOK,TS val 35654980 ecr 0,nop,wscale 7], length 0
13:41:09.332735 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.44965 > ber01s14-in-x04.1e100.net.https: UDP, length 228
13:41:09.333863 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:41:09.334245 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:41:09.337145 IP6 nextnode > 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2: ICMP6, packet too big, mtu 1312, length 1240
13:41:09.341732 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.42079 > ber01s14-in-x04.1e100.net.https: UDP, length 1350
13:41:09.342031 IP6 nextnode > 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2: ICMP6, packet too big, mtu 1312, length 1240
13:41:09.501017 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59670 > ber01s14-in-x04.1e100.net.https: Flags [S], seq 3106298234, win 28800, options [mss 1440,sackOK,TS val 35655022 ecr 0,nop,wscale 7], length 0
13:41:09.652875 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59674 > ber01s14-in-x04.1e100.net.https: Flags [S], seq 412087977, win 28800, options [mss 1440,sackOK,TS val 35655060 ecr 0,nop,wscale 7], length 0
13:41:11.795110 IP6 fe80::9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor solicitation, who has fe80::1441:95ff:fe40:f7dc, length 32
13:41:11.795357 IP6 fe80::1441:95ff:fe40:f7dc > fe80::9eeb:e8ff:fe20:92a2: ICMP6, neighbor advertisement, tgt is fe80::1441:95ff:fe40:f7dc, length 24
13:41:11.852737 IP6 fe80::1441:95ff:fe40:f7dc > fe80::9eeb:e8ff:fe20:92a2: ICMP6, neighbor solicitation, who has fe80::9eeb:e8ff:fe20:92a2, length 32
13:41:11.853123 IP6 fe80::9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is fe80::9eeb:e8ff:fe20:92a2, length 24
13:41:11.991197 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:41:11.991613 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:41:15.702525 IP6 ber01s14-in-x04.1e100.net.https > 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59668: Flags [S.], seq 45665851, ack 1628829022, win 27960, options [mss 1410,sackOK,TS val 2581628569 ecr 35654980,nop,wscale 7], length 0
13:41:15.703023 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59668 > ber01s14-in-x04.1e100.net.https: Flags [R], seq 1628829022, win 0, length 0
13:41:15.872143 IP6 ber01s14-in-x04.1e100.net.https > 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59670: Flags [S.], seq 252324407, ack 3106298235, win 27960, options [mss 1410,sackOK,TS val 2754297122 ecr 35655022,nop,wscale 7], length 0
13:41:15.872692 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59670 > ber01s14-in-x04.1e100.net.https: Flags [R], seq 3106298235, win 0, length 0
13:41:16.025444 IP6 ber01s14-in-x04.1e100.net.https > 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59674: Flags [S.], seq 1318652282, ack 412087978, win 27960, options [mss 1410,sackOK,TS val 2668017736 ecr 35655060,nop,wscale 7], length 0
13:41:16.025805 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.59674 > ber01s14-in-x04.1e100.net.https: Flags [R], seq 412087978, win 0, length 0
13:41:16.992036 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:41:16.992526 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
[... cut ...]
13:43:07.017480 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:43:07.017869 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:43:12.018288 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
13:43:12.018685 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 > fe80::1441:95ff:fe40:f7dc: ICMP6, neighbor advertisement, tgt is 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 32
`

christf, 2016-12-03:
and again same situation (connecting a client to the switch) with igmp_snooping=0 on br-client:

` ./tcpdump  -p -i br-client
tcpdump: WARNING: br-client: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-client, link-type EN10MB (Ethernet), capture size 65535 bytes
[ 1963.740000] eth0: link up (1000Mbps/Full duplex)
[ 1963.740000] br-client: port 1(eth0) entered forwarding state
[ 1963.750000] br-client: port 1(eth0) entered forwarding state
13:46:58.941963 IP6 :: > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:46:59.181976 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:46:59.185618 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::2: ICMP6, router solicitation, length 8
13:46:59.189935 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:46:59.285123 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] PTR (QM)? _ipps._tcp.local. PTR (QM)? _ipp._tcp.local. (45)
13:46:59.321928 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:46:59.357903 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
13:46:59.372440 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [3q] [5n] ANY (QM)? yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local. ANY (QM)? yoshi.local. ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. (257)
13:46:59.397816 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 PTR _workstation._tcp.local., PTR yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local. (112)
[ 1965.750000] br-client: port 1(eth0) entered forwarding state
13:46:59.623599 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [3q] [5n] ANY (QM)? yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local. ANY (QM)? yoshi.local. ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. (257)
13:46:59.649560 IP6 fe80::1441:95ff:fe40:f7dc > ff02::1: ICMP6, router advertisement, length 80
13:46:59.657930 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
13:46:59.890649 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [3q] [5n] ANY (QM)? yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local. ANY (QM)? yoshi.local. ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. (257)
13:47:00.091346 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 5/0/0 (Cache flush) TXT "", (Cache flush) AAAA fe80::9eeb:e8ff:fe20:92a2, (Cache flush) PTR yoshi.local., (Cache flush) HINFO, (Cache flush) SRV yoshi.local.:9 0 0 (239)
13:47:00.245934 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
13:47:00.285744 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] PTR (QM)? _ipps._tcp.local. PTR (QM)? _ipp._tcp.local. (45)
13:47:00.534266 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 PTR _workstation._tcp.local., PTR yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local. (112)
13:47:00.614039 IP6 :: > ff02::1:ff20:92a2: ICMP6, neighbor solicitation, who has 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2, length 24
13:47:00.726159 IP6 fe80::9eeb:e8ff:fe20:92a2.50829 > nextnode.domain: 4056+ SOA? local. (23)
13:47:00.774271 IP6 nextnode.domain > fe80::9eeb:e8ff:fe20:92a2.50829: 4056 NXDomain 0/1/0 (98)
13:47:01.228537 IP6 fe80::9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 5/0/0 (Cache flush) TXT "", (Cache flush) AAAA fe80::9eeb:e8ff:fe20:92a2, (Cache flush) PTR yoshi.local., (Cache flush) HINFO, (Cache flush) SRV yoshi.local.:9 0 0 (239)
13:47:01.616685 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 (Cache flush) PTR yoshi.local., (Cache flush) AAAA fe80::9eeb:e8ff:fe20:92a2 (137)
13:47:01.621894 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:47:01.773913 IP6 fe80::9eeb:e8ff:fe20:92a2 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
13:47:01.803170 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] [2n] ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.2.0.0.0.b.a.b.f.7.8.1.8.6.0.a.2.ip6.arpa. ANY (QM)? yoshi.local. (149)
13:47:01.803316 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 1/0/0 (Cache flush) HINFO (48)
13:47:02.053820 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] [2n] ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.2.0.0.0.b.a.b.f.7.8.1.8.6.0.a.2.ip6.arpa. ANY (QM)? yoshi.local. (149)
13:47:02.286665 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] PTR (QM)? _ipps._tcp.local. PTR (QM)? _ipp._tcp.local. (45)
13:47:02.304929 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0 [2q] [2n] ANY (QM)? 2.a.2.9.0.2.e.f.f.f.8.e.b.e.e.9.2.0.0.0.b.a.b.f.7.8.1.8.6.0.a.2.ip6.arpa. ANY (QM)? yoshi.local. (149)
13:47:02.305054 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 1/0/0 (Cache flush) HINFO (48)
13:47:02.505576 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 (Cache flush) PTR yoshi.local., (Cache flush) AAAA 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 (137)
[... cut ...] p.local., PTR yoshi [9c:eb:e8:20:92:a2]._workstation._tcp.local., (Cache flush) TXT "", (Cache flush) SRV yoshi.local.:9 0 0 (151)
13:47:03.365466 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 4/0/0 (Cache flush) TXT "", (Cache flush) HINFO, (Cache flush) SRV yoshi.local.:9 0 0, (Cache flush) AAAA 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2 (153)
13:47:03.642959 IP6 2a06:8187:fbab:2:9eeb:e8ff:fe20:92a2.mdns > ff02::fb.mdns: 0*- [0q] 1/0/0 (Cache flush) PTR yoshi.local. (109)
`

christf, 2016-12-04:
this time with snooping=1 and full packet capture on eth0 and br-client.
mind: these are captures, not png files.

[image]
[image]

christf, 2016-12-20:
[image]
[image]

christf, 2016-12-20:
[image]
[image]

christf, 2016-12-21:
apparently snooping=1 and multicast_router=2 also results in clients obtaining IP addresses quickly so th is what I am setting in the babel PR for improved airtime.

christf, 2017-09-03:
@T-X what should we do with this?

rotanid, 2019-02-15:
its been another 17 months, whats the status here @T-X @christf  ?

mweinelt, 2020-03-03:
Another 12.5 months … @christf?

closed 2024-01-07 by blocktrron