freifunk-gluon/gluon #960: Linux ipv6 stack: STAs not re-issuing MLD reports on reconnect

T-X, 2016-12-01:
Just to document a known issue with the Linux kernel (afaik all versions as of writing affected):

https://lists.open-mesh.org/pipermail/b.a.t.m.a.n/2015-September/013663.html

The workaround currently implemented in Gluon is to simply have a smaller than default querier interval (4d00c804c0988bc9). But this should eventually be fixed upstream in the Linux kernel, too.

christf, 2016-12-03:
@T-X is this workaround also relevant to babel based networks?

T-X, 2017-02-13:
A patch and proper fix for the Linux IPv6 stack was submitted, got accepted and will be part of 4.10. It should then trickle down to stable kernel releases.

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=a088d1d73a4bcfd7bc482f8d08375b9b665dc3e5

T-X, 2017-02-13:
@christf: Sorry, missed your question back then (but maybe you got an answer via our IRC discussions already?). Otherwise, for the record: As far as I know the babel/l3roamd integration uses a br-client, too. So yes, it is relevant to that as well then.

rotanid, 2017-03-12:
status: not in changelogs of LTS kernels 4.4 or 4.9 yet

T-X, 2017-06-26:
Greg eventually pulled them now. They are part of the 4.4/4.9 LTS kernels since [v4.4.73](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=v4.4.73&id=8d228758f90f234d1c5c6696960a5bd6a392c751) and [v4.9.33](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=v4.9.33&id=1cadd394bbf94976bcffeab54e94f1d083e2d390).

Therefore, I'm closing this ticket now.

closed 2017-06-26 by T-X

rotanid, 2017-06-26:
@T-X shouldn't the issue stay open until we have the updated kernel in gluon and removed the workaround?

T-X, 2017-06-26:
@rotanid: This is a patch for client devices, not nodes.

I had basically opened this ticket back then here to remind myself that I'd need to make a proper patch for the Linux kernel. And to document the issue for Gluon users.

I'd like to close it now because it should eventually trickle to devices now. The workaround with fast query intervals will certainly need to remain in Gluon for a couple more years until this patch makes it into the majority of Android devices. However, I don't think it makes sense to further drag this ticket along until that has happened.