freifunk-gluon/gluon #961: Add support for TL-WR1043nd-v4

thepaffy, 2016-12-02:
One of our community told me that he bought a TL-WR1043nd-v4.
But the v4 is not supported by gluon and, as I see, not by OpenWRT.

So you can add support for the TL-WR1043nd-v4?

rotanid, 2016-12-02:
support can't be added without someone who owns a device and is willing to work on the support, which includes attaching a serial console to the device.
i ordered a 1043 on amazon, hope it will be a v4, but i can't promise how fast the support will be added.

Adorfer, 2016-12-02:
as always where there is new hardware which could possibly be support: Get one unit as donation to the gluon/LEDE-team. 
Other options are available, but they will take much longer and will cause more efforts for you testing build, feeding back logs and reports etc.

rotanid, 2016-12-04:
status update:
we received a 1043nd v4 yesterday and worked the whole night on supporting it, me and @Brother-Lal .
though we are pretty new to adding support for new devices, got some help from @H4ndl3 and are now waiting for @NeoRaider to help with the final touches.
this would result in LEDE support for the device, a backport to gluon would ideally follow.

FabianCernota, 2016-12-17:
Any updates? We got many v4's in our community

Brother-Lal, 2016-12-18:
Support for LEDE is in staging.
After it is main, a backport has to be done for gluon.

rotanid, 2016-12-20:
update: thanks to @Brother-Lal and @neoraider the support for the device is now in LEDE.

commit d9959fae (2016-12-21): ar71xx-generic: backport support for TP-LINK TL-WR1043ND v4

Fixes #961
commit f00082bc (2016-12-21): ar71xx-generic: backport support for TP-LINK TL-WR1043ND v4

Fixes #961
commit bee999dd (2016-12-21): ar71xx-generic: backport support for TP-LINK TL-WR1043ND v4

Fixes #961
PR #991 (2016-12-22): ar71xx-generic: backport support for TP-LINK TL-WR1043ND v4
Fixes #961
closed 2016-12-22 by NeoRaider