freifunk-gluon/gluon #963: Compiling Error

parmenides, 2016-12-04:
Hey, I got the following error after running **$make GLUON_TARGET=ar71xx-generic V=s**
on version (v2016.2.1)
====================
Configuring hostapd-mini.
Configuring gluon-ebtables-filter-ra-dhcp.
Configuring gluon-ebtables-filter-multicast.
Configuring gluon-config-mode-mesh-vpn.
/home/payam/gluon/gluon/Makefile:410: recipe for target 'package_install' failed
make[4]: *** [package_install] Error 255
make[4]: Leaving directory '/home/payam/gluon/gluon/build/ar71xx-generic/openwrt'
/home/payam/gluon/gluon/Makefile:433: recipe for target 'image' failed
make[3]: *** [image] Error 2
make[3]: Leaving directory '/home/payam/gluon/gluon/build/ar71xx-generic/openwrt'
/home/payam/gluon/gluon/Makefile:477: recipe for target 'image/CPE510' failed
make[2]: *** [image/CPE510] Error 2
make[2]: Leaving directory '/home/payam/gluon/gluon/build/ar71xx-generic/openwrt'
/home/payam/gluon/gluon/Makefile:480: recipe for target 'call_image/CPE510' failed
make[1]: *** [call_image/CPE510] Error 2
make[1]: Leaving directory '/home/payam/gluon/gluon/build/ar71xx-generic/openwrt'
Makefile:71: recipe for target 'all' failed
make: *** [all] Error 2

NeoRaider, 2016-12-04:
The log you posted doesn't contain the error, please post a complete log.

parmenides, 2016-12-05:
Sorry, I think i found the error
```
/home/payam/gluon/gluon/openwrt/staging_dir/host/bin/lua: /home/payam/gluon/gluon/scripts/check_site_lib.lua:60: site.conf error: expected `next_node.ip4' to match pattern `^%d+.%d+.%d+.%d+$'
```

and my next_node.ip4 config in site.conf is:
```

  -- Prefixes used within the mesh.
  -- prefix6 is required, prefix4 can be omitted if next_node.ip4
  -- is not set.
  prefix4 = '10.111.111.0/18',
  prefix6 = 'fdca::ffee:babe:1::/64',

```

kb-light, 2016-12-05:
There seems to be something wrong with your next_node IPv4. Please check your site.conf or paste a link here.

closed 2016-12-11 by NeoRaider