freifunk-gluon/gluon #969: docs/user/site.html and docs/features/roles.html differ

txt-file, 2016-12-07:
in docs/user/site.html I should add roles with role name in site.conf and role description in i18n/en.po
```
roles = {
  default = 'node',
  list = {
    'node',
    'test',
    'backbone',
    'service',
  },
},
```

in docs/features/roles.html I should add roles with role name and description inside the site.conf
```
roles = {
  default = 'node',
  list = {
    node = 'Normal Node',
    test = 'Test Node',
    backbone = 'Backbone Node',
    service = 'Service Node',
  },
},
```
Which way is the correct way?

NeoRaider, 2016-12-07:
The first way is correct, I've updated the outdated docs.

closed 2016-12-07 by NeoRaider

txt-file, 2016-12-07:
You are awesome @NeoRaider. I did not expect such a quick reply to this issue.