freifunk-gluon/gluon #973: lede/openwrt does not compile with OpenSSL 1.1

smoe, 2016-12-09:
Hello,

I had accidentially updated the ssl-dev package to 1.1 today and ran into the issue described at
https://github.com/openssl/openssl/issues/1491
This holds for the compilation of
host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c
and then 
<pre>
/usr/include/openssl/err.h:246:1: note: declared here
 DEPRECATEDIN_1_1_0(void ERR_remove_thread_state(void *))
 ^
In file included from tools/lib/rsa/rsa-sign.c:1:0:
/home/moeller/gluon/lede/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function ‘rsa_sign_with_key’:
/home/moeller/gluon/lede/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:213:2: warning: implicit declaration of function ‘EVP_MD_CTX_cleanup’ [-Wimplicit-function-declaration]
  EVP_MD_CTX_cleanup(context);
  ^~~~~~~~~~~~~~~~~~
/home/moeller/gluon/lede/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function ‘rsa_get_exponent’:
/home/moeller/gluon/lede/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21: error: dereferencing pointer to incomplete type ‘RSA {aka struct rsa_st}’
  if (BN_num_bits(key->e) > 64)
                     ^~
scripts/Makefile.host:134: recipe for target 'tools/lib/rsa/rsa-sign.o' failed
</pre>
and so I downgraded libssl again.  I happily test patches but cannot provide them.

Many thanks

Steffen

smoe, 2016-12-09:
I should have said that the above was with the current lede branch and the same was observed with 0.6.2.1 and 0.6.2 while the latter has compiled on the same machine  prior to the update of libssl.

NeoRaider, 2016-12-09:
This has been solved in current LEDE upstream, so the issue will disappear the next time I update the Gluon *lede* branch.

closed 2016-12-09 by NeoRaider

smoe, 2016-12-09:
For others on Debian with the need to downgrade openssl please look at snapshot.debian.org which is an archive of packages that appeared on Debian's servers in the past. Your /etc/apt/sources.list file may then have an extra line like
<pre>
deb   http://snapshot.debian.org/archive/debian/20160927T044906Z unstable main
</pre>
Even then perform the update as in
<pre>
  sudo apt-get -o Acquire::Check-Valid-Until=false update
  sudo apt-get install libssl-dev=1.0.2j-1 libssl1.0.2=1.0.2j-1
</pre>

To learn about the versions available check out 
   apt-cache show libssl-dev|grep Version
   Version: 1.1.0c-2
   Version: 1.0.2j-1
   Version: 1.0.2h-1

NeoRaider, 2016-12-09:
Debian provides the libssl1.0-dev package for this purpose.

smoe, 2016-12-09:
Sigh. Yes. Thank you.

Lotus253, 2017-08-28:
LEDE hasn't it problem

bradley-evans, 2017-08-29:
I did the same thing today. Installing libssl1.0-dev corrected the compilation error.

dankamongmen, 2024-12-30:
you could also just run `./configure --disable-openssl`, especially since this openssl code is old and terrible.