freifunk-gluon/gluon #975: CPE210(EU) v1.1 not recognized

smoe, 2016-12-10:
Hello,

We (Tim in particular) are currently preparing 24 CPE210 for Timmendorf, of which 6 happen to be of an EU flavour. I have no idea about what change to expect internally for those, anyway, this model is not seen as compatible at the time the firmware is meant to be flashed to it.

In extension of https://github.com/freifunk-gluon/gluon/issues/526 , please consider to follow the advice on https://forum.openwrt.org/viewtopic.php?id=64748 and augment the patch [0066-tools-tplink-safeloader-split-CPE210-from-CPE510-profile.patch](https://github.com/freifunk-gluon/gluon/blob/095583ba9221b94eb8b29aabfacd65f39a77de18/patches/openwrt/0066-tools-tplink-safeloader-split-CPE210-from-CPE510-profile.patch)
for the the regions
"CPE210(TP-LINK|US|N300-2):1.1\r\n"
"CPE210(TP-LINK|EU|N300-2):1.1\r\n"
"CPE210(TP-LINK|CN|N300-2):1.1\r\n"
after the line featuring
"CPE210(TP-LINK|UN|N300-2):1.1\r\n"

Many thanks

Steffen

rotanid, 2016-12-10:
there's probably no hardware change, only some firmware change which needs a change adding some kind of region code, which was introduced to many TP-Link devices in the last months.

you didn't really specify what the problem is though, although i guess the firmware upgrade rejects the gluon factory image?

smoe, 2016-12-10:
Yes, even though I have not seen the rejection with my own eyes this is how I understood it. 
We could offer pictures of the board of an EU device if that is any (info|confi)rmative.

Marwe, 2016-12-11:
I have the same issue here with model `CPE210(EU) Ver:1.1`: firmware is rejected on upgrade (Error box: Failure: The hardware version is not supported.)
I upgraded stock firmware from 1.3.3 Build 20160526 Rel. 4923 to 1.3.3 Build 20160705 Rel. 52453 with no difference.

`strings firmware.bin | grep CPE210` shows only `CPE210(TP-LINK|UN|N300-2):1.1`

smoe, 2016-12-11:
@Marwe, nice to hear, many thanks.  I also got the confirmation that this is the exact error we get.
@rotanid, judging "nach Aktenlage" the proposed patch referenced in the initial comment seems just right. Is there much of a point to create a pull request for that? If so, would you have suggestions on how should it be done as this is a patch to a patch? And would the manual change of the patch not disturb the interplay between git and quilt?

Marwe, 2016-12-11:
I guess we need just a compiled version with the patch, to sysupgrade afterwards to the version we want, since it is just the stock firmware having these checks/needing a patched gluon?

I am still compiling gluon, it feels like it will not finish today (OK, it is my first time). And I need to look into how to apply the patch properly. If someone has a clue how to find a binary version with patch applied, please share here.

Marwe, 2016-12-11:
I just edited the string in the binary, but unfortunately that does not work. There may be a CRC check somewhere, since the error message changed to:

> The uploaded file is not correct.

rotanid, 2016-12-11:
if you're lucky, it shouldn't be too much work for @NeoRaider as soon as he has time for it

NeoRaider, 2016-12-11:
Please test, then I'll apply the patches to v2016.2.x.

Also check if `/tmp/sysinfo/model` still contains the correct model name after my changes.

closed 2016-12-11 by NeoRaider

Marwe, 2016-12-11:
Finally I was able to build an image with patches 83 and 84 applied, and the new strings are now in the .bin

> strings gluon-ffs-0.8+*-tp-link-cpe210-v1.1.bin | grep CPE210 
> CPE210(TP-LINK|UN|N300-2):1.0
> CPE210(TP-LINK|UN|N300-2):1.1
> CPE210(TP-LINK|US|N300-2):1.1
> CPE210(TP-LINK|EU|N300-2):1.1

but unfortunately the error box still appears (without error message, just the box with Failure title).
Well, maybe I need to look more into compilation and if the steps were correct. Next days, if I find time.

Marwe, 2016-12-11:
The filename was just too long, so I renamed it and then it worked.
Many thanks, just got into the config...

Message is then
> The new firmware version is 0.0.0 Build 20161211 Rel. 49389. 
> Do you want to keep your configurations or restore to factory default after the upgrade?

Thanks a lot!

NeoRaider, 2016-12-11:
@Marwe, did you also have a look at `/tmp/sysinfo/model`?

Marwe, 2016-12-11:
ssh root@cpe210 cat /tmp/sysinfo/model

> TP-Link CPE210 v1.1

commit 77c649e2 (2016-12-11): ar71xx-generic: update support list for CPE210/510/...

Fixes #975
commit 1e0bc1b4 (2016-12-12): ar71xx-generic: update support list for CPE210/510/...

Fixes #975