freifunk-gluon/gluon #983: make altitude independent of latitude and longitude

FFS-Roland, 2016-12-18:
There are situations, where the exact location (latitude / longitude) should not be published. Because at Freifunk-Stuttgart we use geolocation for assigning nodes to segments (subnets), we want to use altitude  field to store ZIP-Code while latitude / longitude may remain empty. Right now this is not possible because hard coding in "/package/gluon-config-mode-geo-location/luasrc/lib/gluon/config-mode/wizard/0400-geo-location.lua".

To be more flexible in using field altitude, i propose the change to "wizard/0400-geo-location.lua", which is attached.

[patch-0400-geo-location.txt](https://github.com/freifunk-gluon/gluon/files/659664/patch-0400-geo-location.txt)

mweinelt, 2016-12-18:
Please create a new package for your feature, take a look at the `gluon-config-mode-contact-info` as an example: https://github.com/freifunk-gluon/gluon/tree/master/package/gluon-config-mode-contact-info.

This can then be given a dedicted language string and other communities, who'd like to use both, actually can.

FFS-Roland, 2016-12-18:
OK, we will try. Thanx.

closed 2016-12-18 by FFS-Roland

Philhil, 2016-12-25:
If there is somebody in future who also need it. We made a config-mode-module: https://github.com/freifunk-stuttgart/Gluon-Packages/tree/master/gluon-config-mode-zip