freifunk-gluon/gluon #985: Use english as fallback if no german language in browser

rubo77, 2016-12-19:
if you add "en-us" and "de to your browser languages and remove the plain "en" language from your browser laguages, then it doesn't find the "en" translation and takes the next available, which is "de" although "en-us" includes english fine.

We have to change the language selection so it takes only the first part "en" of a flavoured language like "en-us".

NeoRaider, 2016-12-19:
I've tried with multiple browsers and all send an "unflavoured" Accept-language header by default. Not doing so seems to me like an uncommon client misconfiguration we don't need to handle in any way.

Note that we already fall back to English when no matching language is found, so the title of this ticket doesn't make sense either.

closed 2016-12-19 by NeoRaider

rubo77, 2016-12-19:
I was unlucky in formulating the title, corrected it now. But if it really is uncommon not to have an "unflavoured" language, then I am fine with having this closed.