freifunk-gluon/gluon #986: statuspage internationalization

rubo77, 2016-12-19:
For Freemesh Denmark, we will translate the statuspage js files to english (and later danish):

 https://github.com/freifunk-gluon/gluon/tree/master/package/gluon-status-page/src/js/lib/gui

Is there anything planned on adding language support to the statuspage? Please lets collect the Ideas how to implement this properly here in this issue

NeoRaider, 2016-12-21:
Yes, this should be done. No, we don't have a plan yet.

Marlow925, 2016-12-21:
Actually .. we do .. I'm preparing the Freemesh Ireland web server, gateway and images currently, so it will be translated as part of that process.

mweinelt, 2016-12-22:
@Marlow925 The statuspage currently does not offer i18n support, this was only implemented for the config mode so far.

You'll only be able to translate things for which gluon already exposes language strings.

Marlow925, 2016-12-22:
Well .. that at least tells me what to do for now, to get it to english .... meaning having to fork gluon and edit the source before compiling.

rotanid, 2016-12-22:
or you could start working on i18n support for it ;)

Marlow925, 2016-12-22:
I'm no software developer. Fairly straight-forward customization, translation and basic changes, I can handle. I'm a network guy.

And trust me .. you would not appreciate the code, I would add painfully to the project.

rubo77, 2016-12-22:
Don't fork gluon, just copy the status page packet and create a new statistics English packet that can be included in the sites modules file

rubo77, 2016-12-30:
I created a package `gluon-status-page-en` that you can add in your module and site.mk:

https://github.com/Freemesh-Ireland/site-fmie/commit/a6a2e7f5a2042154614e32e260a5d4e20810dc7f

mweinelt, 2016-12-31:
The issue with a fork like this is that it won't automatically be updated from the project upstream, as it would if language strings were added.

Marlow925, 2016-12-31:
Agreed, but it's a working shortcut way of getting an english status page for now. 

Internationalisation seems to be a bit of a problem, as the strings are embedded in the javascript code.

rubo77, 2017-01-04:
actually there are three files to be translated. Here the diffs: https://github.com/freifunk-kiel/ffki-packages/compare/ddfd37e7ae6b47d1547fa6b7721a847d7b57befa...308c62b5515132ee4c3325acbfc3b1ad683e7200

I will try to add a simple translation functionality with this script: http://stackoverflow.com/questions/41338417/create-a-lua-like-javascript-i18n-internationalization/41459139#41459139

rubo77, 2017-01-04:
I started the internationalisation here: https://github.com/rubo77/gluon/tree/status-i18n/package/gluon-status-page

**I need some help now,** how to implement the `gettext-example.js` into the existing code.

norn, 2017-01-05:
I would like to propose another way. Declare translate function and cover all strings into it. This looks simpler, work out of the box and can be done in main repository without forking. 

```javascript
function _(s) {
     var lang = navigator.language || navigator.userLanguage
     var dict = {
      "en-US": {
        "Knoten": "Node",
        "Entfernung": "Distance",
        "Inaktiv": "Inactive"
      },
      "ru-RU": {
        "Knoten": "Узел",
        "Entfernung": "Distance",
        "Inaktiv": "Inactive"
      }
     }
     if (lang in dict && s in dict[lang])
       return dict[lang][s]
     else
       return s
}
// and then "Knoten" becomes _("Knoten")
```

rubo77, 2017-01-05:
The simplest and possible without forking already is to include the englich module from https://github.com/freifunk-kiel/ffki-packages/tree/master/gluon-status-page-en

If this is wanted, I create a PR to include the package gluon-status-page-en into standard gluon?

Maybe this is the best solution, cause an **english-only** replacement version would serve most countries already and uses the least precious extra space on our nodes

NeoRaider, 2017-01-06:
@norn's solution is much better than duplicating the package. The default language should be English though, in case no matching language is found (and is there a nice way to access the whole list of languages that is sent in the Accept-language header from JavaScript?).

norn, 2017-01-08:
Chrome and Firefox support [navigator.languages](https://developer.mozilla.org/en-US/docs/Web/API/NavigatorLanguage/languages).

PR #1044 (2017-02-21): gluon-status-page: JS translations
Fixes #986
closed 2017-02-21 by NeoRaider