freifunk-gluon/gluon #988: v2016.2.2 build failure - no gluon-next-node

viisauksena, 2016-12-20:
as i understood gluon-next-node is now in gluon-core, but if i try to build actual master or last tag v2016.2.2 fw build failed at this point 

...
make[4]: Entering directory '/media/freifunk/firmware/gluon/build/ar71xx-generic/openwrt'
make[4]: *** package/feeds/gluon/gluon-next-node: No such file or directory.  Stop.
...

a dirty quickfix was to give back the initial gluon-next-node source directory, by just copying from v2016.2.1 tag.
there was no difference if site.mk includes gluon-next-node or not, but maybe i miss something essential.

NeoRaider, 2016-12-20:
Maybe you forgot a `make update` when switching branches or something similar?

v2016.2.2 is a maintenance release and obviously still has gluon-next-node (and still needs it in site.mk); we never require site changes in maintenance releases. Only the Gluon master has no gluon-next-node package anymore.

closed 2016-12-20 by NeoRaider

viisauksena, 2016-12-20:
mhm, than i have to search for it, maybe a pull of latest gluon is hidden somewhere