freifunk-gluon/gluon #993: All Meraki MR16 have the same mac-address

Marlow925, 2016-12-25:
Hi,
all Meraki MR16 have the same mac-address, after they're flashed with OpenWRT or for that sake Gluon.

LEDE have already found a fix for this. Is there a way to either integrate this patch into Gluon ?
Or is the move to LEDE so far progressed, that we can sort it that way ?

http://lists.infradead.org/pipermail/lede-dev/2016-October/003459.html

Marlow925, 2016-12-28:
Found a temporary workaround by editing /etc/config/network or set the appropriate mac-addresses using uci.

A permanent solution, like what LEDE has done would be better though, so that it doesn't get reset when all data is reset.

Marlow925, 2016-12-28:
Nevermind .. that still doesn't work for gluon. Gluon takes the physical mac-address as nodeid and eth0 address, no matter what has been set by uci or /etc/config/network.

The first MR16 added to the network will work perfectly. The second one will end up in a bootloop, once mesh is enabled, causing kernel dumps until it reboots as there's already another node on the network.

NeoRaider, 2017-01-09:
Backporting the solution in LEDE to Gluon would be simple, but I really don't like it. Instead of using the MAC partition of the stock firmware, they define a new partition, so it is still necessary to touch each device by hand to assign MAC addresses. This hacky solution was chosen to allow upgrades from older OpenWrt/LEDE versions without the "fix"...

I'll mark the Meraki devices as BROKEN for now, till we have a proper solution. If we really want to invest time in this, that is, as these devices are crappy overpriced hardware that should just disappear...

Marlow925, 2017-01-09:
I agree, that without the LEDE approach or another way of preserving the MAC address, they're broken.

I wouldn't consider them overpriced though, as they're a very decent priced, when bought second hand. Nobody would buy them new though with only wanting to flash them with OpenWRT/LEDE/Gluon.

Hardware wise, they're really nice piece of kit.

marecki12345, 2018-02-12:
Hi everyone.
Urgent !!
Need dump of flash (mx25l12845e) from Meraki mr16. Any dump with working boot sector. Mac address don't mater for me(will install openwrt ). In my mr16 has been Accidentally overwritten.

rotanid, 2019-02-15:
closing this issue together with #1004  as it has to be fixed upstream and we cant track every tiny upstream problem for non-supported hardware.

closed 2019-02-15 by rotanid