freifunk-gluon/gluon #994: REQUEST: add Fonera 2.0n (FON2303A) to Gluon

Marlow925, 2016-12-25:
Hi,
a substantial donation has been made in the form of routers to some of the FreiFunk groups. 

Of these, at least 1200 routers are Fonera 2.0n (FON2303A). Description here: http://www.freemesh.ie/wiki/index.php/Fonera_FON2303A

Can we please get this added to the mainstream Gluon projekt ?

All that is needed is:

gluon/targets/ramips-rt305x/profiles.mk

+
+ $(eval $(call GluonProfile,FONERA20N))
+ $(eval $(call GluonProfileFactorySuffix,FONERA20N))
+ $(eval $(call GluonModel,FONERA20N,fonera20n,fonera-fon2303a))

The firmware will build without problems. Of course, these routers will only run 802.11s, due to the Ralink chipset, but the run Freifunk stable, as what we can see from tests in 4 regions (tested with firmware for Freifunk Flensburg, Freifunk Kiel, Freemesh Denmark and Freemesh Ireland) Freifunk Brauschweig also received a few, but i'm unsure if they've deployed them yet. They're also using Gluon.

There's about 1200+ of these routers, that will be distributed to Freifunk projects and this is also the only Fonera Router, that made it into LEDE so far, so I figured, that would justify getting it added to Gluon.

rotanid, 2016-12-26:
if that is really all that is needed and you have tested everything, feel free to open a pull request to add the device to gluon.
keep in mind that you have to put it there using the BROKEN flag as it doesn't support ibss

Marlow925, 2016-12-26:
Yes, I'm aware, that the BROKEN flag still needs to be set on compile, but apart from IBSS not being a possibility, these routers actually work quite nicely and will offer a very cheap entry point into Freifunk.

I'll see, how I create a Pull Request. Thanks. It will at least remove the need from patching the Gluon tree to compile these images.

rubo77, 2016-12-27:
I think, this can now be closed by https://github.com/freifunk-gluon/gluon/pull/995

Marlow925, 2016-12-27:
Jup

closed 2016-12-27 by Marlow925