freifunk-gluon/gluon #1008: Master branch build fails to download ath10k firmware

smoe, 2017-01-20:
Dear all,
I was eager to try the LEDE branch and after a git pull did the "make update ; make clean" without problems. Then, with several GLUON_TARGETs, the default make target fails to perform a download. Here the one for x64_64. 
The latest commit is
```
commit 70b116fd619a9956b5761957c358441c41604146
Author: Matthias Schiffer <mschiffer@universe-factory.net>
Date:   Thu Jan 19 18:00:46 2017 +0100

    build: apply model-specific site packages at the end of the package list
```

Cheers,
Steffen
```
if [ -f /home/moeller/gluon/lede/staging_dir/target-x86_64_musl-1.1.15/pkginfo/ath10k-firmware.default.install.clean ]; then rm -f /home/moeller/gluon/lede/staging_dir/target-x86_64_musl-1.1.15/pkginfo/ath10k-firmware.default.install /home/moeller/gluon/lede/staging_dir/target-x86_64_musl-1.1.15/pkginfo/ath10k-firmware.default.install.clean; fi
mkdir -p /home/moeller/gluon/lede/dl
SHELL= flock /home/moeller/gluon/lede/tmp/.board-2.bin.deb1832c56c706d0f6cb539113e09f0daaa52b5f.flock -c '  	/home/moeller/gluon/lede/scripts/download.pl "/home/moeller/gluon/lede/dl" "board-2.bin.deb1832c56c706d0f6cb539113e09f0daaa52b5f" "6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e" "board-2.bin?id=deb1832c56c706d0f6cb539113e09f0daaa52b5f" "https://source.codeaurora.org/quic/qsdk/oss/firmware/ath10k-firmware/plain/ath10k/QCA9984/hw1.0/"    '
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Connection timed out after 20001 milliseconds
Download failed.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168  100   168    0     0    660      0 --:--:-- --:--:-- --:--:--   661
MD5 sum of the downloaded file does not match (file: 6b8b4de583fe80567ab1fc585465793a957ae6ddbb7a3e2740f6621ef583997b, requested: 6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e) - deleting download.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168  100   168    0     0  12023      0 --:--:-- --:--:-- --:--:-- 12923
MD5 sum of the downloaded file does not match (file: 4adb44b3cd6fe503d218067307302628c3a0a895acfe03998c24c8f3d561dd15, requested: 6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e) - deleting download.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   162  100   162    0     0   2512      0 --:--:-- --:--:-- --:--:--  2531
MD5 sum of the downloaded file does not match (file: f33c27745f2bd87344be790465ef984a972fd539dc83bd4f61d4242c607ef1ee, requested: 6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e) - deleting download.
No more mirrors to try - giving up.
Makefile:193: recipe for target '/home/moeller/gluon/lede/dl/board-2.bin.deb1832c56c706d0f6cb539113e09f0daaa52b5f' failed
make[4]: *** [/home/moeller/gluon/lede/dl/board-2.bin.deb1832c56c706d0f6cb539113e09f0daaa52b5f] Error 2
make[4]: Leaving directory '/home/moeller/gluon/lede/package/firmware/ath10k-firmware'
package/Makefile:105: recipe for target 'package/firmware/ath10k-firmware/compile' failed
make[3]: *** [package/firmware/ath10k-firmware/compile] Error 2
make[3]: Leaving directory '/home/moeller/gluon/lede'
package/Makefile:101: recipe for target '/home/moeller/gluon/lede/staging_dir/target-x86_64_musl-1.1.15/stamp/.package_compile' failed
make[2]: *** [/home/moeller/gluon/lede/staging_dir/target-x86_64_musl-1.1.15/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/moeller/gluon/lede'
/home/moeller/gluon/lede/include/toplevel.mk:197: recipe for target 'world' failed
make[1]: *** [world] Error 2
make[1]: Leaving directory '/home/moeller/gluon/lede'
Makefile:120: recipe for target 'all' failed
make: *** [all] Error 2
```

NeoRaider, 2017-01-20:
Works for me. "Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.": Either the server was down a moment ago, or there is some routing issue that makes it unreachable for you. In any case, this is not a problem in Gluon or LEDE.

closed 2017-01-20 by NeoRaider

smoe, 2017-01-20:
Hi Matthias, thank you for your swift reply. I presume it is a bit more difficult since I had tried the whole night and could just now repeat the problem, now a new prestine checkout with no earlier builds of leda. All other downloads work and the server can be addressed with the web browser:
```
de/staging_dir/host/bin/patchelf" /home/moeller/gluon/lede/scripts/rstrip.sh /home/moeller/gluon/lede/build_dir/target-mips_24kc_musl-1.1.15/linux-ar71xx_generic/nat46-7/ipkg-mips_24kc/kmod-nat46
rstrip.sh: /home/moeller/gluon/lede/build_dir/target-mips_24kc_musl-1.1.15/linux-ar71xx_generic/nat46-7/ipkg-mips_24kc/kmod-nat46/lib/modules/4.4.42/nat46.ko: relocatable
(cd /home/moeller/gluon/lede/build_dir/target-mips_24kc_musl-1.1.15/linux-ar71xx_generic/nat46-7/ipkg-mips_24kc/kmod-nat46/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/moeller/gluon/lede/bin/targets/ar71xx/generic/packages
/home/moeller/gluon/lede/scripts/ipkg-build -c -o 0 -g 0 /home/moeller/gluon/lede/build_dir/target-mips_24kc_musl-1.1.15/linux-ar71xx_generic/nat46-7/ipkg-mips_24kc/kmod-nat46 /home/moeller/gluon/lede/bin/targets/ar71xx/generic/packages
Packaged contents of /home/moeller/gluon/lede/build_dir/target-mips_24kc_musl-1.1.15/linux-ar71xx_generic/nat46-7/ipkg-mips_24kc/kmod-nat46 into /home/moeller/gluon/lede/bin/targets/ar71xx/generic/packages/kmod-nat46_4.4.42+7-1_mips_24kc.ipk
rm -rf /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46
mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/stamp /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46
. /home/moeller/gluon/lede/include/shell.sh; export modules=; probe_module() { mods="$1"; boot="$2"; shift 2; for mod in $mods; do mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d; echo "$mod" >> /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d/nat46; done; if [ -e /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d/nat46 ]; then if [ "$boot" = "1" ]; then mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules-boot.d; ln -s ../modules.d/nat46 /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules-boot.d/; fi; modules="${modules:+$modules }$mods"; fi; }; add_module() { priority="$1"; mods="$2"; boot="$3"; shift 3; for mod in $mods; do mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d; echo "$mod" >> /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d/$priority-nat46; done; if [ -e /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d/$priority-nat46 ]; then if [ "$boot" = "1" ]; then mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules-boot.d; ln -s ../modules.d/$priority-nat46 /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules-boot.d/; fi; modules="${modules:+$modules }$priority-nat46"; fi; }; add_module "33" "nat46" ""; if [ -n "$modules" ]; then mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/etc/modules.d; mkdir -p /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL; echo "#!/bin/sh" > /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL/postinst-pkg; echo "[ -z \"\$IPKG_INSTROOT\" ] || exit 0" >> /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL/postinst-pkg; echo ". /lib/functions.sh" >> /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL/postinst-pkg; echo "insert_modules $modules" >> /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL/postinst-pkg; chmod 0755 /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/CONTROL/postinst-pkg; fi
SHELL= flock /home/moeller/gluon/lede/tmp/.root-copy.flock -c 'cp -fpR /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46/. /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/'
rm -rf /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/tmp-kmod-nat46
touch /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/stamp/.kmod-nat46_installed
if [ -f /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/pkginfo/nat46.default.install.clean ]; then rm -f /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.15/pkginfo/nat46.default.install /home/moeller/gluon/lede/staging_dir/target-mips_24kc_musl-1.1 [... cut ...]
https://source.codeaurora.org/quic/qsdk/oss/firmware/ath10k-firmware/plain/ath10k/QCA99X0/hw2.0/board-2.bin?id=ddcec9efd245da9365c474f513a855a55f3ac7fe
which was saved as board-2.bin and has the md5 
 $ md5 board-2.bin
MD5 (board-2.bin) = a2b3c653c2363a5641200051d6333d0a

Tried both Linux and Mac, browser and wget, pasted to file and (as in the download.pl script) piped through, toyed with the LANG settings. Always that same hash. This does not match the hash value that is expected (but admittedly neither the hash value that it claims to have found).

Anybody else sharing that experience?

Best,

Steffen

smoe, 2017-01-20:
Ah, the requested hash comes up not as an md5 but as a sha256sum, not as an md5, so I got an issue with the error message at least that does not take up he distinction from line 59 in  lede/scripts/download.pl 
```
sub hash_cmd() {
        my $len = length($file_hash);
        my $cmd;

        $len == 64 and return "mkhash sha256";
        $len == 32 and return "mkhash md5";
        return undef;
}
```
Let me see. Sounds like something that I address. There is something.

NeoRaider, 2017-01-20:
LEDE doesn't use MD5 hashes anymore. The expected SHA256 hash of the file is `6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e`, and this is exactly the hash I see when I manually download https://source.codeaurora.org/quic/qsdk/oss/firmware/ath10k-firmware/plain/ath10k/QCA9984/hw1.0/board-2.bin?id=deb1832c56c706d0f6cb539113e09f0daaa52b5f

smoe, 2017-01-20:
This is also what I just got, pasting your URL to the command line after a wget -O - . However, the hash sum you request is 03711ac21e60ef59d3815e235eb721c0c22851b5410299411085aa6f2af45401 and it is for a different version, i.e. ddcec9efd245da9365c474f513a855a55f3ac7fe .

I would be glad to see something along
```
-                       print STDERR "MD5 sum of the downloaded file does not match (file: $sum, requested: $file_hash) - deleting download.\n";
-                       cleanup();
+                       my $len = length($file_hash);
+                       if    ($len == 32) {
+                               print STDERR "MD5 sum of the downloaded file does not match (file: $sum, requested: $file_hash) - deleting download.\n";
+                       }
+                       elsif ($len == 64) {
+                               print STDERR "SHA256 sum of the downloaded file does not match (file: $sum, requested: $file_hash) - deleting download.\n";
+                       }
+                       else {
+                               print STDERR "Error: Unexpected length of requested Hash: $len\n";
+                               exit(1);
+                       }
```
corrected - I tried on MD5 for too long. In my local version I have commented out the cleanup and just checked the .dl file:
```
<html>
<head><title>404 Not Found</title></head>
<body bgcolor="white">
<center><h1>404 Not Found</h1></center>
<hr><center>nginx</center>
</body>
</html>
```
which explains why there are different checksums with every attempt. I need to install some Freifunk routers now (pre-Lede :o/ ) and will be back on this in two hours.

NeoRaider, 2017-01-20:
`ddcec9efd245da9365c474f513a855a55f3ac7fe` is the commit ID, `6a79ff0e8cc71549e771b41dbb7dad862d8e29da852f8aff25ce1e4bd5ea263e` is the hash of the file. Completely different things.

smoe, 2017-01-20:
There was just an "of" missing in my descriptive text, sorry for that. I got it to work. For the wanted connection to source.codeaurora.org the firewall was too strict. For my reporting above I had sadly focused my investigation on the later lines on mirrors that do not accept the ?id= specification and returned file-not-found pages. 

So, you were right in that the problem was with me. And I was right that there is something problematic, which is that the mirrors cannot function because of different URL construction rules. Thank you for your help and feedback. If there is something you want me to work on in that area, like for functional mirrors, then tell me.

Best,

Steffen

NeoRaider, 2017-01-20:
Ah, the ?id= failure for the other mirrors is interesting, I'll try to get this fixed in LEDE.