freifunk-gluon/gluon #1009: build of air71xx-nand failed in attempt to cp something -generic

smoe, 2017-01-22:
Hello,
I ran the air71xx-generic, mpc85xx-generic and x64-generic targets successfully, but with ar71xx-nand or ar71xx-tiny the very last action in the prepare image target had difficulties:
```
Generating index for package ./luci-lib-nixio_git-17.014.75996-d73283606-1_mips_24kc.ipk
Generating index for package ./ecdsautils_0.3.2.20160630-1_mips_24kc.ipk
Generating index for package ./eudev_3.2-1_mips_24kc.ipk
Generating index for package ./fastd_18-2_mips_24kc.ipk
Generating index for package ./haveged_1.9.1-5_mips_24kc.ipk
Generating index for package ./libecdsautil_0.3.2.20160630-1_mips_24kc.ipk
Generating index for package ./libhavege_1.9.1-5_mips_24kc.ipk
Generating index for package ./libudev_3.2-1_mips_24kc.ipk
Generating index for package ./libuecc_7-2_mips_24kc.ipk
Generating index for package ./libwrap_7.6-1_mips_24kc.ipk
Generating index for package ./alfred_2016.4-0_mips_24kc.ipk
Generating index for package ./batctl_2016.4-0_mips_24kc.ipk
make[3]: Leaving directory '/home/moeller/gluon/lede'
export MAKEFLAGS= ;make -w -r diffconfig
make[3]: Entering directory '/home/moeller/gluon/lede'
make[3]: Leaving directory '/home/moeller/gluon/lede'
export MAKEFLAGS= ;make -w -r checksum
make[3]: Entering directory '/home/moeller/gluon/lede'
make[3]: Leaving directory '/home/moeller/gluon/lede'
make[2]: Leaving directory '/home/moeller/gluon/lede'
make[1]: Leaving directory '/home/moeller/gluon/lede'
cp: cannot stat 'lede/bin/targets/ar71xx/generic/lede-ar71xx-generic-dir-615-c1-squashfs-factory.bin': No such file or directory
Makefile:120: recipe for target 'all' failed
```

I investigated a bit further and came to the conclusion that the toplevel Makefile is not specifying the subboard incorrectly. It is always set to 'generic'. And these varlues are then passed to the copy script, which then (luckily) fails.

I did this patch to the Makefile to inform me

```
--- a/Makefile
+++ b/Makefile
@@ -59,6 +59,7 @@ LEDEMAKE = $(MAKE) -C lede
 
 BOARD := $(GLUON_TARGET_$(GLUON_TARGET)_BOARD)
 SUBTARGET := $(GLUON_TARGET_$(GLUON_TARGET)_SUBTARGET)
+#SUBTARGET := tiny
 LEDE_TARGET := $(BOARD)$(if $(SUBTARGET),-$(SUBTARGET))
 
 export LEDE_TARGET
@@ -121,7 +122,9 @@ all: prepare-target
                 || (echo 'Your site configuration did not pass validation.'; false)
 
        @scripts/clean_output.sh
+       @echo "I: (Makefile) BOARD=$(BOARD) SUBTARGET=$(SUBTARGET) GLUON_TARGET=$(GLUON_TARGET) LEDE_TARGET=$(LEDE_TARGET)"
        +@$(LEDEMAKE)
+       @echo "I: (Makefile) GLUON_TARGET=$(GLUON_TARGET) LEDE_TARGET=$(LEDE_TARGET)"
        @GLUON_SITEDIR='$(GLUON_SITEDIR)' scripts/copy_output.sh '$(GLUON_TARGET)'
 
 clean download: config
```
which gave
I: (Makefile) BOARD=ar71xx SUBTARGET=generic GLUON_TARGET=ar71xx-tiny LEDE_TARGET=ar71xx-generic

Many thanks and greetings

Steffen

NeoRaider, 2017-01-22:
This is a common error, it happens when the images are too big. I'm currently working on reducing the size of the LuCI packages we use. Till that is done, you can reduce the image size be removing additional languages (set `GLUON_LANGS := en` only).

smoe, 2017-01-22:
```
I: (Makefile) GLUON_LANGS='en' BOARD=ar71xx SUBTARGET=generic GLUON_TARGET=ar71xx-tiny LEDE_TARGET=ar71xx-generic
```

No, this is not it. I suspect it something dead simple in the Makefile or its include files that I just fail to see.

smoe, 2017-01-22:
No idea why, but it seems to work now. The prior report on tiny was unfortunate since the platform is indeed the generic one, just a tiny flavour of it.

closed 2017-01-22 by smoe