freifunk-gluon/gluon #1010: x86-64: sysupgrade (soft)fails

kpanic23, 2017-01-23:
When upgrading to master, sysupgrade writes the image and stops. After manual reboot, config is lost, device boots to config mode.

I have tried upgrading from 2016.2.2 to (aaa4150) --> fail.
After newly configuring the device downgrade from master to 2016.2.2 works flawlessly.
Upgrade from 2016.2.2 to last master before LEDE PR (04818c1) --> works.
Upgrade from (04818c1) to (aaa4150) --> fail.
After newly configuring the device, upgrade from (aaa4150) to same (aaa4150) --> fail
So everything seems to be OK until the LEDE PR.

Here's the output from sysupgrade, maybe it could help:

```
root@hoho-ho:/tmp# wget http://ftp.ff3l.net/master/output/hoho/experimental/sysu
pgrade/gluon-ff3l-hoho-master~20170119-x86-64-sysupgrade.img.gz
Downloading 'http://ftp.ff3l.net/master/output/hoho/experimental/sysupgrade/gluon-ff3l-hoho-master~20170119-x86-64-sysupgrade.img.gz'
Connecting to 2a01:4f8:121:3145::2:80
Writing to 'gluon-ff3l-hoho-master~20170119-x86-64-sysupgrade.img.gz'
gluon-ff3l-hoho-mast 100% |*******************************|  6980k  0:00:00 ETA
Download completed (7148183 bytes)
root@hoho-ho:/tmp# sysupgrade gluon-ff3l-hoho-master~20170119-x86-64-sysupgrade.
img.gz 
Image metadata not found
Saving config files...
Sending TERM to remaining processes ... logd haveged netifd uradvd micrond sse-multiplexd uhttpd dnsmasq fastd ntpd dnsmasq respondd odhcp6c ubusd askfirst 
Sending KILL to remaining processes ... askfirst 
Switching to ramdisk...
Performing system upgrade...
Reading partition table from bootdisk...
0+1 records in
0+1 records out
Reading partition table from image...
Writing image to /dev/sda1...
32768+0 records in
16+0 records out
Writing image to /dev/sda2...
packet_write_wait: Connection to fdc7:3c9d:ff31:1:2e0:fcff:fe68:320a port 22: Broken pipe
```

NeoRaider, 2017-01-24:
Unfortunately, I can't reproduce the issue here, neither with images I built myself, nor with the one from your log. It might be a hardware-specific issue, I'm using VMware for x86 tests.

The log of the sysupgrade looks sane.

kpanic23, 2017-01-25:
Hm, strange...
I have been testing on a Wyse R90LEW thin client (we have started using them as offloaders instead of the Futro S550)
Is it somehow possible to debug the sysupgrade process? I'm quite curious at exactly which point the upgrading borks out before the reboot (which does not happen)...

MTRNord, 2017-01-25:
Happend to me also as I used the img on a Oracle VM. But my VM is not on the newest master. (It is at 054b1094ffe6e30f457a1842326767108f57ebb1 )

NeoRaider, 2017-01-25:
Might be affected by this: https://git.lede-project.org/?p=source.git;a=commit;h=b02c381b4b817e22a36ce5b2155e9eebcc77e312 , I'll update to the latest LEDE-17.01 version soon.

rotanid, 2017-02-12:
as @NeoRaider updated the master branch to latest lede, please check again if the issue persists @MTRNord @kpanic23

kpanic23, 2017-02-15:
Testing done:
```
root@Frhlingswyse:/tmp# wget http://devrepo.ff3l/hoho/lede/sysupgrade/gluon-ff3l
-hoho-master~20170214valentine-x86-64-sysupgrade.img.gz
Connecting to devrepo.ff3l ([fdc7:3c9d:ff31:1:250:43ff:fe01:6772]:80)
gluon-ff3l-hoho-mast 100% |*******************************|  6999k  0:00:00 ETA
root@Frhlingswyse:/tmp# sysupgrade gluon-ff3l-hoho-master~20170214valentine-x86-
64-sysupgrade.img.gz 
Saving config files...
Sending TERM to remaining processes ... logd haveged netifd uradvd micrond sse-multiplexd uhttpd dnsmasq fastd ntpd dnsmasq respondd ubusd askfirst askfirst 
Sending KILL to remaining processes ... askfirst askfirst 
Switching to ramdisk...
Performing system upgrade...
```
Then after quite some time:
```
69760+0 records in
69760+0 records out
mount: mounting /dev/sda1 on /mnt failed: Invalid argument
umount: can't umount /mnt: Invalid argument
Upgrade completed
Rebooting system...
```
Now the device reboots in Setup Mode, configuration is lost.

NeoRaider, 2017-02-27:
@kpanic23 It would be great to get a kernel log from the failed mounts. Maybe you could patch the *sysupgrade* script to add a *dmesg* before the *reboot*?

NeoRaider, 2017-04-09:
Upstream ticket: https://bugs.lede-project.org/index.php?do=details&task_id=685

NeoRaider, 2017-06-01:
See 88f2e3e4e2de3666705cc4bc4198c728e7fa4ebe

NeoRaider, 2017-06-02:
Staged sysupgrade patches have been backported to v2016.2.x in d4a69c00047f72696a2400cd7129be032de458e3. This will need some testing before we release v2016.2.6, but I didn't find any problems on ar71xx or x86.

closed 2017-06-02 by NeoRaider

A-Kasper, 2017-06-02:
works for me.
I've installed 2016.2.5 x86-gen on futro 550s
then I made a sysupgrade to 2016.2.x latest commit
then I made a sysupgrade to latest master
everything worked fine.

SimJoSt, 2017-06-12:
Updated WDR4900 and CPE210 from 2016.2.6 to 2016.26 to test the new sysupgrade procedure.
Worked like a charm. I quite like the notice messages, even though it's less information. Best possible way.

kpanic23, 2017-06-12:
Erm... neither WDR4900 nor CPE210 are x86 ;)

NeoRaider, 2017-06-12:
@kpanic23: the sysupgrade has been more or less rewritten for all targets, so testing is necessary everywhere.

commit fcce9e52 (2017-05-30): generic: revert to OpenWrt CC partition sizes on x86 and similar images for now

This will allow upgrading using the broken sysupgrade scripts in OpenWrt
CC without losing the configuration.

Fixes #1010