freifunk-gluon/gluon #1011: Set BROKEN flag for  tl-wr1043nd v1

rubo77, 2017-01-23:
We had a lot of issues with the old v1 of this device.

please set a broken flag for this device, since it really doesn't work round with gluon

T-X, 2017-01-23:
Can you provide a ticket describing how to reproduce the issues with a tl-wr1043nd v1? Have lots of 'em here. And what the issues are, actually? Is it the LEDs blinking morse code saying "we will enslave all humans" or is something else broken? ;-)

heini66, 2017-01-23:
+ one for me

Adorfer, 2017-01-24:
the BROKEN will not help the fact that's probably the WiFi Chipset which is not stable when running IBBS and AP simultaneously. 
And since everyone is aware of the shortcomings of these units, it was the same since AA...

Brother-Lal, 2017-01-30:
There have been some fixes In the current Lede build regarding the ath9k wifi drivers, see https://github.com/lede-project/source/commit/f9022964cfa0a44984d2efa2dafe3452899acfdf and https://github.com/lede-project/source/commit/36db14369016b976ce7b0d6a1928eb7a26c45f5c

I think after this the 1043 might be usable again. Its still old hardware, though...

Adorfer, 2017-02-01:
@Brother-Lal 
> I think after this the 1043 might be usable again. Its still old hardware, though...

O.k. let's stay optimistic, but honestly i do not have much confidence that any patches will help 1043v1, it has been suffereing from "unreliable" wifi not only in CC, but BB and AA was not different. At least what i observed. 

In other words: People with this device know what they are dealing with, since they presumingly use the device for years now. 
Setting a BROKEN mark will not prevent anyone from purchasing a "new" one, nor will it help any user of an existing installation. 
It will just kill the autoupgrade upgrade path in domains where firmware is not compiled with "BROKEN" Flavour plus pushing it in the stable-branch. 

(if a domain is by autoupgrade changing fastdservers, batman-versions etc it will kick those units out...)

And as disclaimer: I myself know how to get around that issues, but here will be probably some communities not so heavily involved in gluon-buildscripts which will get in trouble, whilst nobody will benefit from a newly instroduced "BROKEN" on 1043V1.

closed 2017-02-01 by rotanid