freifunk-gluon/gluon #1020: DIR825B1: Images >4 MB are too big

mweinelt, 2017-01-31:
This is an upstream issue in LEDE.

Building the ar71xx-generic target fails, when the image grows past 4 MB.
```
Warning: /scratch/hexa/build/gluon/lede/bin/targets/ar71xx/generic/lede-ar71xx-generic-dir-825-b1-squashfs-sysupgrade.bin is too big
```

The DIR825b1 has 8 MB of Flash but is limited through the CameoAP94 profile it uses, which specifies a maximum image size here:
https://git.lede-project.org/?p=source.git;a=blob;f=target/linux/ar71xx/image/legacy.mk;h=d2f05e993eab55e461ef6cb7a8f465ccd08a4cb0;hb=refs/heads/master#l334

mweinelt, 2017-02-09:
I created an upstream issue at https://bugs.lede-project.org/index.php?do=details&task_id=485.

NeoRaider, 2017-02-23:
I've had a look at this, and the factory image is always exactly 4MB (the sysupgrade image may be bigger though). I guess we have two options:

* move the DIR825B1 to ar71xx-tiny
* make DIR825B1 sysupgrade-only, requiring to install a plain LEDE image before.

SimJoSt, 2017-02-23:
Just out of curiosity and as I don't know anything about this:
Why would making it sysupgrade-only decrease the size of the sysupgrade image?
Isn't using the factory image like installing a (modded and extended) LEDE already?
What would change?

NeoRaider, 2017-02-23:
@SimJoSt The error message @mweinelt pasted is generated when the sysupgrade image is converted to factory image format. So the image is actually fine to be used for sysupgrades.

closed 2017-02-24 by NeoRaider

A-Kasper, 2017-02-24:
> Closed #1020 <https://github.com/freifunk-gluon/gluon/issues/1020> via
> 0af0be0

I don't like this solution. I don't want to say users to install lede
first. I would prefer to add a factory image "prepare" with minimum
packages where users could use the known config mode to sysupgrade.

The package list in site.mk should be ignored for those images. instead a
defined "prepare image only" set of packages should be installed. the
imagename should get a suffix -prepare-only.

maybe longterm a autocheck of image sizes would be possible.
communitypackages could increase the sizes:

if imagesize is in limit use it
else if sysupgradesize is in limit make prepare image for factory
else try to build tiny version
if this still is to big throw error "too many packages for device"

A-Kasper, 2017-02-24:
maybe it's possible to modify config mode for prepare images to show the fw
update dialogue with an extra message only?

A-Kasper, 2017-02-24:
After thinking about it, this would really be the best solution.
Communities wouldn't need to explain the installprocedere for these devices
and could just upload the image in their factory image folder as usual.
after installing it the user would be guided through the modify configmode
"sysupgrade only" to upload the sysupgrade image.

maybe it would also make sense to check autoupdateurl for sysupgradeimage
automatic and download and install it in some way and just show modified
config mode only where this fails...

commit 0af0be06 (2017-02-24): ar71xx-generic: don't try to build a factory image for D-Link DIR-825 B1

The size of the factory images is limited to 4MB, which caused build
failures when many additional packages were included.

Rather than moving the device to ar71xx-tiny, we ignore the factory images
and just build the sysupgrade. This way, the whole flash is usable for
Gluon.

This means that installing Gluon on these devices will now require to flash
a plain LEDE image first, and then upgrading using the Gluon sysupgrade
image.

Fixes #1020