freifunk-gluon/gluon #1022: Unterstützung für Netzwerkkarte Realtek 8139C+ in x86-generic

ChristophFranzen, 2017-02-02:
Hallo, ich bekomme folgendes im Dmesg: `[  101.556169] 8139too 0000:02:0d.0: This (id 10ec:8139 rev 20) is an enhanced 8139C+ chip, use 8139cp`.

Hardware ist eine „Watchguard Firebox X-Core“, die ich für das VPN nehmen möchte.

Ist es möglich, den „8139cp.ko“ standardmäßig zu integrieren?

rotanid, 2017-02-02:
basically it would be possible to include it, but if your really need it, just include it yourself via the site.mk file.
if you're in no hurry, you may create a pull request adding this kmod to the affected targets.

ChristophFranzen, 2017-02-03:
Hallo Andreas, danke für dioe Rückmeldung, ich denke hier eher langfristig, also „in a hurry“ trifft es nicht ganz, so daß ich – obgleich ich das schon mal erfolgreich geschafft habe – jetzt keinen „pull request“ generiere. Der Punkt hier: ältere Realteks werden unterstützt durch „8139too.ko“, neuere auch (8169), aber dazwischen klafft diese Lücke, obwohl der Chip nicht wirklich selten ist, eher im Gegenteil. Der ist wohl auch weniger für AMD-64 relevant, weil nicht Gigabit-fähig, aber der generische i386 sollte den Treiber drin haben, denke ich, sonst fehlt für die Hardware-Generation was. Ich bin auch für Eigenleistung nicht zu faul, damit Du das jetzt nicht falsch verstehst: die erbringe ich bei der LCD-Anzeige, den LEDs und der Crypto-Karte von den Watchguards, das alles gehört nicht in ein Standard-Paket, weil zu selten und speziell. Sollte sich doch noch herausstellen, daß es diesbezüglich mehr Interessenten gibt, wäre ein „x86-Watchguard-Firebox“-Image angesagt.

commit 8ea481ed (2017-02-03): x86: add kmod-8139cp by default

Fixes #1022
closed 2017-02-03 by NeoRaider