freifunk-gluon/gluon #1023: CPE 510 still low tx power

mokkin, 2017-02-03:
The CPE 510 tx problem still exists, even in my first try with LEDE-based gluon. dB is about 20 dbm less.
I'm still using the old firmware at this point: https://github.com/freifunk-gluon/gluon/tree/ffd1f0b3a5b2950d8c626a2de26fb1dc8e384408

NeoRaider, 2017-02-03:
The last update on this matter was https://git.lede-project.org/?p=source.git;a=commit;h=c2e0c41842895ba47819fa98b785c76a2524628b .

mokkin, 2017-02-06:
I think it was reverted due to usage of an upper channel. @dangowrt could you add your experience here?
On lower (indoor) channels (36-48) with less tx power there is absolutely no problem and we have a couple of CPE510 working fine and stable since month ago.

2tata, 2017-03-11:
I can confirm this problem. I have some CPE510 in combination with CPE210 and all clients tend to use cpe210 because the CPE510 tx power seems very low. state is v2016.2.x.

mokkin, 2017-03-14:
I manually used the higher power table with an actual LEDE based gluon like NeoRaider suggest and it worked like before. If you stay on the lower 5GHz channels (36-48) it should work without any damage.

FreifunkUFO, 2018-02-19:
@mokkin we could do a gluon- PR/patch!? and include it at least to our gluon-repo

mokkin, 2018-03-25:
@FreifunkUFO Yes. I think we should include it at least to our https://github.com/freifunk-leipzig/freifunk-gluon-leipzig as there aren't as many 5 GHz users around here.
@NeoRaider Or could we revert the patch for the lower indoor channels (36-48)?

NeoRaider, 2018-03-25:
This must be fixed in OpenWrt. If a fix is not acceptable for OpenWrt, it is not for Gluon.

Note that #1329 will allow to enable DFS channels on many 5GHz devices by disabling meshing. This will become the default on all outdoor 5GHz devices supported by Gluon, including the CPE510.

rotanid, 2018-11-25:
this upstream issue will be tracked in Gluon over there: #1156
closing.

closed 2018-11-25 by rotanid

mokkin, 2021-12-23:
Do you have any idea how can I revert the patch in the actual gluon master or OpenWRT 21.02? Until 19.07 I can find the postion of the file to revert it, but since migration to ath79 it has changed and I couldn't find it anymore.

NeoRaider, 2021-12-24:
@mokkin The calibration data offset is now defined here: https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ath79/dts/ar9344_tplink_cpe.dtsi;h=9d8873ddaf1b3232aee81e59f3f522ac9b3a9b73;hb=HEAD#l86

The following should work to override it only for the CPE510 v1:

```diff
diff --git a/target/linux/ath79/dts/ar9344_tplink_cpe510-v1.dts b/target/linux/ath79/dts/ar9344_tplink_cpe510-v1.dts
index b1fe2364cd65..b728f255f137 100644
--- a/target/linux/ath79/dts/ar9344_tplink_cpe510-v1.dts
+++ b/target/linux/ath79/dts/ar9344_tplink_cpe510-v1.dts
@@ -10,3 +10,7 @@
 &led_link4 {
 	gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
 };
+
+&wmac {
+	mtd-cal-data = <&art 0x5000>;
+};
```

mokkin, 2022-01-15:
> @mokkin The calibration data offset is now defined here: [...]

Thanks a lot for you patch @NeoRaider !

Those threee lines worked perefectly on my recent builds with OpenWRT and TX power is back. I guess they will work with gluon as well.

ddreamer75, 2022-07-15:
is it possible to change settings on running gluon fw ?