freifunk-gluon/gluon #1024: gluon-config-mode-geo-location: unable to leave Long/Lat empty

T-X, 2017-02-04:
Using current master, I'm having some trouble with leaving the geo position undefined.

I have left "Show node on the map" unticked and left Long/Lat empty then. However, the wizard then complains with a "One or more required fields have no value!":

[image]

The browser I've tried is Midori 0.4.3 on an ARM based system.

rubo77, 2017-02-04:
Ia this reproducible?

Maybe you entered a space somewhere?

And is js enabled?

NeoRaider, 2017-02-04:
Anything in your Js console? The problematic fields should not be shown at all when "Show node on the map" is unset.

T-X, 2017-02-05:
Hrm, it was reproduceable for the second device.

Now I had rebuilt Gluon with a more recent version of LEDE, restarted midori with logging - and the issue is gone. Probably some browser caching issue, I guess *sigh*.

closed 2017-02-05 by T-X