freifunk-gluon/gluon #1025: LEDE: bridge multicast snooping is disabled

T-X, 2017-02-04:
With the switch to LEDE bridge multicast snooping was disabled, because two people were reporting issues with multicast.

http://lists.infradead.org/pipermail/lede-dev/2017-January/005482.html

For one of the two, the issues were confirmed to lie somewhere else, not within the bridge. For the other one, investigations are still pending:

https://bugs.lede-project.org/index.php?do=details&task_id=253

I did some quick retries and for me, the bridge multicast snooping seems to be working as expected, just like it did with OpenWRT.

For testing, I installed the gluon-ebtables-segment-mld package and added "option igmp_snooping '1'" and "option multicast_querier '1'" to the client section in /etc/config/network.

I marked this issue as a regression for now, as there might already be people relying on a reduced multicast level in neighborhoods with many, close nodes.

NeoRaider, 2017-04-11:
@T-X Can you prepare a PR to enable multicast snooping again?

T-X, 2017-04-11:
Sure, will do today or tomorrow.

I haven't heard of any complains with 2016.2.x even together with the gluon-ebtables-segment-mld package so far and the LEDE bug report [253](https://bugs.lede-project.org/index.php?do=details&task_id=253) mentioned above progresses very slowly, is more confusing than helpful so far and therefore is still unconfirmed. So I think it is safe to reenable it, yes.

PR #1096 (2017-04-18): LEDE: reenable multicast snooping
LEDE recently disabled multicast snooping by default:

https://git.lede-project.org/?p=project/netifd.git;a=commitdiff;h=52541140f8138e31958cdc3d7e42a4029fa6bbc9

Reenable it for Gluon as there have been no confirmed issues for
LEDE and no negative reports concerning Gluon v2016.2.x so far.

Closes #1025.
commit 66ead88d (2017-04-18): gluon-client-bridge: reenable multicast snooping for client zone

LEDE recently disabled multicast snooping by default:

https://git.lede-project.org/?p=project/netifd.git;a=commitdiff;h=52541140f8138e31958cdc3d7e42a4029fa6bbc9

Reenable it for Gluon as there have been no confirmed issues for
LEDE and no negative reports concerning Gluon v2016.2.x so far.

Closes #1025.
commit c519ec45 (2017-04-18): gluon-core: reenable multicast snooping for wan zone

LEDE recently disabled multicast snooping by default:

https://git.lede-project.org/?p=project/netifd.git;a=commitdiff;h=52541140f8138e31958cdc3d7e42a4029fa6bbc9

Reenable it for Gluon as there have been no confirmed issues for
LEDE and no negative reports concerning Gluon v2016.2.x so far.

Closes #1025.
closed 2017-04-18 by NeoRaider