freifunk-gluon/gluon #1037: FeatureReq: in config-mode save what's typed in simple mode

DeusFigendi, 2017-02-12:
Today I flashed a range extender and again I run into a failure I always do if I did not flash a router five minutes ago:

I download and upload a new firmware to the device, it reboots and comes up with the config-mode. I type in all my basic data like the hostname, the position, my contact-information and so on.
Then I always switch to the advanced-tab to set up my ssh-keys so I will be able to easy config the device later without physical access. Sometimes I do some more "advanced" settings and then I switch back to the simple settings -tab (or whatever it's called in English) to save & reboot and… all settings are gone :-( And I have to retype everything and that's a little annoying especially about the position.

Here's my suggestion: Save the data of these input-fields in the sessionStorage of the browser so they can easy be restored if the user comes back.

DeusFigendi, 2017-02-12:
Oh of corse it might be also okay to seperate the "save" and the "reboot" button like it works in the advanced mode. So I could type in everything, save that (without rebooting) and then switch to the advanced -tab. So getting back to simple-mode it would load the settings from the device I'd guess.

NeoRaider, 2017-02-12:
Duplicate of #131.

As a workaround, always fill in the advanced settings first before completing the wizard.

closed 2017-02-12 by NeoRaider

DeusFigendi, 2017-02-12:
Ah I didn't found that one, thank you.

DeusFigendi, 2017-02-13:
Yea sure, you know the problem is: I'm not flashing routers every day. And if I didn't for a while I just forget this problem.
So yes, when I flash a dozen devices in a row of corse I found my workaround. But not on the first one.

@rubo77 that idea also was mentioned on #131 but it might be usefull if I remember this problem after I filled the form but before I hit the advanced-tab :-)

Adorfer, 2017-02-14:
Perhaps it would be helpful to make the link to the "andvanced" tab in "simple" to open a new tab by default.