freifunk-gluon/gluon #1038: respondd does not answer queries to [ff02::1]:1001 anymore

mweinelt, 2017-02-12:
On https://github.com/freifunk-gluon/gluon/commit/dc2cc47594ea93fb94cf25d953713a2e6e5e0928 respondd does not answer to queries anymore.

```
# tcpdump -ni br-client udp and not port 53
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-client, link-type EN10MB (Ethernet), capture size 262144 bytes
06:50:42.959366 IP6 fe80::d8ff:61ff:fe00:1104.45123 > ff02::2.1001: UDP, length 14
06:50:48.969203 IP6 fe80::d8ff:61ff:fe00:1104.45123 > ff02::2.1001: UDP, length 14
```

respondd is running:
```
# ps | grep respondd
 3937 root      2156 S    /usr/bin/respondd -d /usr/lib/respondd -d /lib/gluon/respondd -p 1001 -g ff02::2:1001 -i mesh0 -i br-wan -i br-client -t 10 -g ff05::2:1001 -i br-client -t 10
```

I assume this issue was introduced in https://github.com/freifunk-gluon/gluon/pull/1030.

mweinelt, 2017-02-12:
The node in question is:
https://map.darmstadt.freifunk.net/#!v:m;n:e894f6694d86
http://[2001:67c:2ed8:6100:ea94:f6ff:fe69:4d86]/

NeoRaider, 2017-02-13:
Querier was configured to address `ff02::2` instead of `ff02::2:1001`, which incidentially worked before #1030; still, `ff02::2:1001` is the correct address and is working as intended.

closed 2017-02-13 by NeoRaider