freifunk-gluon/gluon #1039: rdnss handling broken

mweinelt, 2017-02-12:
On https://github.com/freifunk-gluon/gluon/commit/dc2cc47594ea93fb94cf25d953713a2e6e5e0928 support for RDNSS is apparently broken.

This is a router advertisment captured on said device:
```
06:59:56.898587 IP6 (flowlabel 0x67d4f, hlim 255, next-header ICMPv6 (58) payload length: 152) fe80::d8ff:61ff:fe00:604 > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 152
	hop limit 64, Flags [none], pref medium, router lifetime 1800s, reachable time 0s, retrans time 0s
	  mtu option (5), length 8 (1):  1280
	    0x0000:  0000 0000 0500
	  prefix info option (3), length 32 (4): 2001:67c:2ed8:6100::/64, Flags [onlink, auto], valid time 86400s, pref. time 14400s
	    0x0000:  40c0 0001 5180 0000 3840 0000 0000 2001
	    0x0010:  067c 2ed8 6100 0000 0000 0000 0000
	  prefix info option (3), length 32 (4): fd01:67c:2ed8:6100::/64, Flags [onlink, auto], valid time 86400s, pref. time 14400s
	    0x0000:  40c0 0001 5180 0000 3840 0000 0000 fd01
	    0x0010:  067c 2ed8 6100 0000 0000 0000 0000
	  rdnss option (25), length 40 (5):  lifetime 1800s, addr: fd01:67c:2ed8:a::55:1 addr: fd01:67c:2ed8:a::56:1
	    0x0000:  0000 0000 0708 fd01 067c 2ed8 000a 0000
	    0x0010:  0000 0055 0001 fd01 067c 2ed8 000a 0000
	    0x0020:  0000 0056 0001
	  dnssl option (31), length 24 (3):  lifetime 1800s, domain(s): ffda.io.
	    0x0000:  0000 0000 0708 0466 6664 6102 696f 0000
	    0x0010:  0000 0000 0000
```

`/tmp/resolv.conf.auto` is empty :(
```
# cat /tmp/resolv.conf.auto 
# Interface client
```

The router correctly acquires an IPv6 address from the announced prefix.
```
# ip addr show dev br-client
11: br-client: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether e8:94:f6:69:4d:86 brd ff:ff:ff:ff:ff:ff
    inet6 fd01:67c:2ed8:6100:ea94:f6ff:fe69:4d86/64 scope global dynamic 
       valid_lft 86399sec preferred_lft 14399sec
    inet6 2001:67c:2ed8:6100:ea94:f6ff:fe69:4d86/64 scope global dynamic 
       valid_lft 86371sec preferred_lft 14371sec
    inet6 fe80::ea94:f6ff:fe69:4d86/64 scope link 
       valid_lft forever preferred_lft forever
```

NeoRaider, 2017-02-13:
The issue was found to be a broken *site.conf*.

closed 2017-02-13 by NeoRaider

mweinelt, 2017-02-13:
Not a bug.

We set `dns.servers` in our `site.conf` to wrong entries and that broke DNS resolution.

When `dns.servers` is set it is wrong to expect the nameservers in `/tmp/resolv.conf.auto`, but in `/etc/config/dhcp` within the `dnsmasq` block.