freifunk-gluon/gluon #1047: Package sizes

rubo77, 2017-02-17:
Is there a list of sizes of each package?

Maybe we should add the size a package uses in the firmware to each package.

Adorfer, 2017-02-17:
1) good idea 
2) no idea how to do that since it's heavily depending on the target-architecture, compiler version an used libs (BB/CC/Lede)

viisauksena, 2017-02-18:
this is "easily" visible by the ipk files - look at this for example to get a rough idea 
http://www.openfreiburg.de/freifunk/firmware/ipk/
or kmod (kernel modules for a recent gluon v2016.2.3)
http://www.openfreiburg.de/freifunk/firmware/modules/gluon-fffr-v2017.1.0.e-support/ar71xx/generic/
but there are plenty of dependencies .. so gre6 depends on gre and ip6tunnel for example, so some packages are mor complex in terms of package size.

rubo77, 2017-02-18:
I think from now on we should add the sizes one after another over time to the readmes

NeoRaider, 2017-02-18:
This will just get outdated often and be more confusing than helpful.

closed 2017-02-18 by NeoRaider

christf, 2017-02-18:
I usually look at the dir list of the openwrt package mirror to determine an estimate for the size of a package. This will always be up to date. 

-- 
Diese Nachricht wurde mit K-9 Mail gesendet.