freifunk-gluon/gluon #1051: einzelne cpe verlieren eth0.1 - br-wan mit 2016.2.3 (und vermutlich davor)

viisauksena, 2017-02-21:
https://forum.freifunk.net/t/phy0-dma-failed-cpe210/10374/27

es ist schwer zu debuggen,[ ein einzelner cpe](http://openfreiburg.de/freifunk/meshi2/#!v:m;n:60e327530852) hat neben anderen 841 an der gleichen box immer wieder das problem das er sein eth0.1 aus dem br-wan fliegt und in folge keinen eigenen uplink im mesh-netz mehr zur verfügung stellen kann. Vereinzelt gibt es eine  dma fehlermeldung im dmesg, aber nicht immer. 
andere CPE an anderen orten haben das Problem nicht, auch wenn sie an Technisch identischen DSL routern mit gleichem DSL Anbietern hängen.
```
brctl addif br-wan eth0.1
udhcpc -i br-wan -b -n
/etc/init.d/fastd restart
```
das ist an diesem Router nun zum 3. mal passiert, und erst stunden nach funktionierenden online aufgetreten.

NeoRaider, 2017-02-22:
Please provide the output of `logread` when this happens. You can increase the size of the log buffer using `uci set 'system.@system[0].log_size=1024'` (or even bigger) if you have trouble catching the error in the logs.

viisauksena, 2017-02-22:
thx, there was nothing special on logread - afaik - or i cant say, cause fastd isspaming quiet a lot - while this uplink router is important, i set it to wireguard uplink - this works fine 
(but this also can be because regularly running script )
all other cpe210v1 and v1.1 seem to be unaffected
(i tried reflashing with fastd before, but this also runs in regular dropping of IF)

even if some mentioned it in the forum, it looks actually more like a less-likely-to-happen scenario (but still buggy) - @Adorfer brought in the idea this could also befactory defects - and because of the big numbers on devices - we see this 
so if not other come forward this is a dead end (just nice to know)

viisauksena, 2017-03-07:
since nobody else mentioned something similar i close this

closed 2017-03-07 by viisauksena