freifunk-gluon/gluon #1054: gluon-status-page, gluon-mesh-babel: extend statistics for babel respondd module

christf, 2017-02-23:
these resources remain unimplemented in the gluon-mesh-babel respondd provider:
    nodeinfo.software.babeld
    statistics.clients
    statistics.traffic
    statistics.gateway
    statistics.gateway_nexthop
    neighbours.*

christf, 2017-04-12:
@NeoRaider how do you feel about turning statistics.gateway into a list contianing the fields ipaddress, txcost and rxcost? I would rather not put the additional two fields for the cost on the same level like gateway itself as they correspond to the gateway only.

NeoRaider, 2017-04-12:
Do we actually know the IP address of the gateway, and not just the nexthop?

Changing this sounds like a good idea, but for consistency, adding the metric (TQ) to the gateway field for batman-adv would make sense, too. I don't think we can do that though without breaking compatiblity unless we introduce a new name for the "gateway" field. Something like "default_route" might make sense.

christf, 2017-04-12:
I had a bad and wrong implementation for obtaining the gateway IP address as I just noticed. 
We do know the fe80-address of gateway_nexthop which I just included in the implementation (build & test still has to happen)

What do we do about the gateway then? I cannot see a field from the babel data from which we could derive that. A traceroute might reveal that. But do we actually want respondd to run traceroute to determine that information?

still it would be good to know which gateway was used....

NeoRaider, 2017-04-12:
In a layer 3 routing protocol, we usually just don't know.

christf, 2017-04-12:
so I guess we should skip the field then.
This would also mean there is no need to rework the json structure for batman for consistency....

=> The issue could be closed then. The implementation is already part of the babel-PR.

closed 2017-04-12 by christf