freifunk-gluon/gluon #1057: Autoupdater sometimes only works with specific mirror

mmalte, 2017-02-26:
Some of our nodes did not take the autoupdate to 2016.2.3, mainly running 2016.1.5. They didn't even load the manifest.

We provide to hosts for autoupdates, the one with a mesh internal, but global, IPv6 provided the manifest all the time. 71 out of 1618 online nodes with stable autoupdater did not load the manifest from this host.

I assumed that this is the known autoupdater bug.

But since today we provide the up to date manifest from both servers, as a result there are onle 14 nodes with an old firmware are left. All of this nodes loaded the update during one hour.

Is there another community with only one out of n autoupdate servers providing the manifest?

mmalte, 2017-02-26:
I just checked the nodes with a pending update: They are shown online in hopglass, but only because there are mesh neighbours receiving their mesh wifi. None of them has responded to respondd in the last hours.

This means all online nodes are up to date now, even ~500 starting from 2016.2.2

Is there another community that is only using some of their autoupdate servers that might confirm this issue?

viisauksena, 2017-03-01:
i thought that it was kinda random ... at least if i start autoupdate manual, i see it trying the same mirro in a row (we have 2 out of 6 invalid - because of reasons ) - and i have to redo it a couple of times. But it works. So maybe it is this , a minor fraction which gets the "wrong" mirror in row which will be fixed by time?

jplitza, 2017-03-01:
Correct. The mirrors of the autoupdater are used in random order, trying the next one *only* if the previous one failed in some way (manifest was corrupted, not for the correct branch, didn't carry enough signatures, …). However, as soon as a valid manifest was found and verified, the autoupdater stops and doesn't look if any other mirror has a newer version. That's why they are called *mirrors*. 😉

So this basically is by design, hence I'm closing the issue. Keep your mirrors in sync or don't use multiple mirrors at all. Using a DNS name instead of an address still allows you to manually switch to another server if the one dies.

(Also, you might be interested in #454, which takes into account which mirror has which version, but is yet vaporware)

closed 2017-03-01 by jplitza

mmalte, 2017-03-07:
Well, the default behaviour is that the mirror is choosen randomly, but I had a considerable number of nodes only checking one out of two mirrors for days.
~5 days after the delay set up in the manifest sums up to over 100 tries, all ending up with randomly choosing the the same mirror?