freifunk-gluon/gluon #1058: bedarf an gluon-mesh-wireguard?

viisauksena, 2017-03-01:
wollen wir ein offizielles gluon-mesh-wireguard ? 
ähnlihc dem mesh-vpn mesh-tunneldigger und mesh-babel ...
ich hab das ziemlich weit voran getrieben, und hier im Testbetrieb hervorragend am laufen.
hier der Fork der tendenzell in Frage kommt
https://github.com/viisauksena/gluon-mesh-wireguard

er liest die site.conf Daten aus (derzeit nur 1 server)
er hat einen config mode eintrag
er generiert key und tunnelinterne ipv6 (und zeigt die am ende des config mode an)
läd nur wen br-wan "echten" uplink hat

NeoRaider, 2017-03-01:
* Please no German in the issue tracker, I might start deleting such tickets. Gluon is not only used in Germany, and the issue tracker should be useful to everyone.
* You're mixing up package types, we have
  * mesh routing protocols (gluon-mesh-batman-adv-*, gluon-mesh-babel)
  * VPN protocols (gluon-mesh-vpn-fastd, gluon-mesh-vpn-tunneldigger)

So a wireguard package would be called "gluon-mesh-vpn-wireguard". We could add such a package, but I don't like the GRE hackery. Let's rather work on adding wireguard support for Babel when that is merged, so we can use the Layer 3 tunnel directly.

viisauksena, 2017-03-07:
thats why i am asking beforehand ;) 
so i think there is no need for a package like this at this moment, or people can use mine directly - and later we can do a clean layer 3 tunnel package with babel (which still is confusing me)

christf, 2017-03-08:
Well, it would be a good idea to start with a clean wireguard l3 
integration now. babel is underway. see #934

viisauksena, 2017-03-25:
@christf , didnt get your last message, you mean we should work on a gluon-mesh-vpn-wireguard ? 
but which purpose should be fullfilled in regard to babel, 
actually i have a little tricky wireguard setup running, but some problems or decissions have to be made, namely
* do we want a l2tp like bridge construction server/gw-side? or how to handle some hundred possible nodes which would like to connect?
* do we want gretap and batman-adv on top, or are we happy with routing at this point
* if we were happy with routing, some decissions in regard to the network/backbone has to be made - and this also has to be built in package

RubenKelevra, 2017-03-27:
Just a side note ...

I already asked the wireguard-author several month ago if he is planning or might consider to add a tap- instead of a tun-mode for wireguard.

He immediately responded, he is not considering to add a tap-mode because he don't think that L2 tunnelling is often used and most of the time unneeded, furthermore he wants to keep it as small as possible and don't blow it up with more stuff which is rarely needed.

So if someone might thought about waiting for a tap-mode to be implemented in wireguard I have to disappoint you, this seem to be not happening.

NeoRaider, 2017-03-27:
@RubenKelevra We're aware of these limitations. 1:n connections are possible in TAP mode only, not in TUN mode, so wireguard can't support it either. The current plan is to add a gluon-mesh-vpn-wireguard module after Babel support is merged (a Layer 3 tunnel should be enough for Babel).

I don't see the need for a fastd kernel module as wireguard exists now.

RubenKelevra, 2017-03-29:
Hey Matthias,

> @RubenKelevra We're aware of these limitations. 1:n connections are
> possible in TAP mode only, not in TUN mode, so wireguard can't
> support it either. The current plan is to add a [...]
But there is not an actual limitation for a L3 tunnel to be only 1:1,
is it?

I think OpenVPN supports L3 tunnels 1:n very fine. So the question is,
why shouldn't wireguard do this as well? :)

Did somebody already ask the author for this feature?

NeoRaider, 2017-03-29:
@RubenKelevra OpenVPN uses ugly hacks, there are only two ways it can work:
1. When only a single address assigned by OpenVPN itself is assigned to a peer
2. When the subnets routed to a specific peer are explicitly configured in the OpenVPN config
Otherwise, OpenVPN can't know what peer a specific IP address lies behind, and "via" routes don't work because they rely on a L2 the "gateway" can reached over.

None of these solutions are compatible with dynamic routing as used in our mesh networks. The proper solution is to use 1:1 tunnels.

RubenKelevra, 2017-03-29:
Hey Matthias,

> @RubenKelevra OpenVPN uses ugly hacks, there are only two ways it can
> work: 1. When only a single address assigned by OpenVPN itself is
> assigned to a peer 2. When the subnets routed to a specific [...]
Yeah, OpenVPN isn't a very good example, because it's way to hacky in
every way.

Since Wireguard uses it's own protocol for communication, wouldn't it
be possible do something similar to dhcp on a connect after
authenfication?

Then Wireguard knows each tunnel-IP pair without the need for that
much interfaces and the scripted setup stuff for each new node.

Because one interface per Freifunk-node doesn't look like a very
favoritable solution to me either. We have over 1200 nodes configured at
the moment, which means we get a lot of interfaces on each server...

Or is there any limitation which would make it impossible for babel to
use this?

NeoRaider, 2017-03-29:
It's really about Babel needing to add arbitrary routes to arbitrary peers. A L3 interface only allows adding interface routes, the "via" address of such routes is ignored completely; this makes it impossible to define the destination peer when multiple peers are connected behind a single interface.

Using 1:1 tunnels is the a way to solve this without hacks (and I don't know why we're even discussing this; 1:1 tunnels are a sane solution as long as there's tooling to automatically establish large numbers of such tunnels, like there is with Tunneldigger for L2TP).

RubenKelevra, 2017-03-30:
> It's really about Babel needing to add arbitrary routes to arbitrary
> peers. A L3 interface only allows adding interface routes, the "via"
> address of such routes is ignored completely; this makes it [...]
Alright, thanks, I had no time to deeply look into Babel. But this
sounds absolutly reasonable. 

Do you have tested such a large scale setup out? Means wireguard
establishing thousands of tunnels while babel is using them? :)

Just don't want to run into trouble if I'm the first to test that out,
while everybody else is just using 200 tunnels per server. :)

viisauksena, 2017-03-30:
on wireguard side there will only be one tunnel iface serverside, which is fine when all the packets are routet. You may dig deeper into wireguard. the handling will be similar to fastd, where 1 "server" has many peers. but all peers mostly only one server.
the actual confusing thing may that some (like me) build wireguard and on top use gretap and batman-adv - to mimic this big layer 2 network, or fastd behaviour. But this wont be the goal of this wireguard packet.

RubenKelevra, 2017-03-31:
@viisauksena thanks for your comment. Well, but if so, we still need 1:1 tunnels for babel if I understood NeoRaider correctly. 

I think I have to dig deeper in both topics, Babel as well as Wireguard, before I ask further stupid questions here. :)

Well, thanks for your headup, I guess I will start to just follow this discussion until I had the time to read both manuals/documentations.

But I want to make one thing clear, I loved the batman-adv approach for the one-net approach, because local devices can find eachother, if we can implement efficient multicast-routing as well as node-based IGMP-snooping and multicast-to-unicast on the wifi-side with Babel itself I'm sure we will change to Babel.

Wireguard just looks good because it have strong crypto and is fast, which is currently a HUGE bottleneck for us. Fastd always worked great for us, so I think I'll miss it ;)

RubenKelevra, 2017-06-04:
@Neoraider

I've talked a bit to a guy who is currently developing babeld. He said that WireGuard is currently incompatible to babeld because babeld needs a multicast-able transport which WireGuard doesn't provide. But support for unicast-only links are in active consideration because they want to implement TLS-security between peers.

Additionally we can use gretap or similar above WireGuard to make it compatible with babeld.

CodeFetch, 2018-01-12:
I have looked at l3roamd and we don't need multicast as it can handle the routes. WireGuard is great (40 MBit/s Batman <-> GRETAP <-> WireGuard). I'm working on getting it supported right after Babel (and also with Babel)...

christf, 2018-05-30:
l3roamd + babeld depend on working multicast for route distribution at the moment. Even with the unicast patchset, not all multicasts by babeld are replaced.

CodeFetch, 2018-05-30:
The WireGuard architecture is very simple. Patching WireGuard to support multicast should be no problem.

christf, 2018-08-18:
I have started work to integrate wireguard as l3 VPN based on the code above. The code will shortly be in my christf_next branch on https://github.com/christf/gluon/tree/christf_next

yanosz, 2018-09-06:
From an IPSec point of view, defining an l2tp pseudowire on top of wireguard appears to be reasonable - just like IPSec offers a a transport-mode.

As an alternative, adding a vxlan layer using unicast-endpoints is consistent with gluon's 802.11s configuration.
```
ip link add vxlan0 type vxlan id <id> dev <wireguard-if> dstport 0  
bridge fdb append to 00:00:00:00:00:00 dst <wg-server> dev vxlan0
```

CodeFetch, 2018-11-13:
@yanosz Before fiddling around with L2TP/VXLAN over WireGuard please consider a fork of WireGuard with L2 support as it is very straightforward. WireGuard can easily be adopted to our needs.

CodeFetch, 2018-11-14:
@kpanic23 If the person considers security and maintainability aspects porting WireGuard to L2, but if he or she just wants a q&d solution porting fastd to kernel space. zx2c4 does not consider adding L2 support until WireGuard is merged into mainstream kernel, but in the long run L2 could be merged. We need more people capable of writing C/kernel code and both quests should be a good starting point.

kpanic23, 2018-11-15:
Maybe a project for the upcoming GSoC?

rotanid, 2018-11-15:
this issue here will be done by #1534  - for babel/L3 meshes.

yanosz, 2019-05-02:
twmic: In the meantime @johnnybee created a wireguard implementation using gretap.
* Package is: https://github.com/johnnybee/gluon-mesh-vpn-wireguard
* Server is: https://github.com/johnnybee/create-supernode-BonnV2

I guess, that this is WIP atm. At KBU's community day in February this year we discussed open todos (docs, e.g. why gretap instead of l2tp/tunneldigger) and I guess, that johnybee is working on that.

CodeFetch, 2019-05-02:
I've a poc implementation of layer 2 WireGuard, but I assume I won't have free-time to finish it until the end of the year and I'd like to get some funding to do it properly. E.g. 4-way-handshake for perfect forward privacy hiding of node's identities and solving the time problem with that. 
With using layer 2 support in WireGuard christf's broker scripts are not needed as Babel can then route the traffic according to link local IPs.

 I also have a WireGuard/GRETAP implementation, but I wasn't happy with the MTU.

almereyda, 2020-07-22:
It would be interesting to hear, how the latest gretap or layer 2 implementations have progressed in favour of #1534 meanwhile.

2tata, 2020-07-22:
We already have VXLAN on the gluon Routers. @NeoRaider @CodeFetch  is that not a way to overcom the missing layer2 support of wireguard?

CodeFetch, 2020-07-22:
@2tata I really don't like such a bodge job. You know... It's not that we are unable to do it right. I have a GRE-WireGuard-Patch lying around here for years, but a tunnel in a tunnel means too much overhead/MTU issues. In my view the right thing to do is to just create a layer 2 tunnel based on WireGuard. Originally I've talked to Jason Donenfeld and he said it was okay to integrate layer 2 support in WireGuard, but then it became upstream and the API is now fixed. Thus we would need to create an own tunnel interface. I know how to do it, but I don't have any time for it. At some point [this](https://github.com/CodeFetch/TunnelBridge/commits/master) was working except some packet length issues which would need further investigation. The hard problem is to bring it upstream.

By all means at the moment I don't think it makes any sense to put effort into WireGuard as lemoer and I are currently evaluating io_uring. io_uring allows to do IO in an asynchronous manner without a lot of context switches. To be clear: We hope for fastd to double its throughput.

CodeFetch, 2020-07-22:
@2tata Do you want to join us in our evaluation? This is the current working state (there is a commit missing for fixing a memory leak): https://github.com/lemoer/fastd/commits/socket
But it only reaches half of the throughput of an ordinary fastd. 

To give you a brief overview of what we are currently examining: 
io_uring has a mode called SQPOLL where a kernel thread takes care of IO submissions leaving us with NO context switch at all. We expected this to be of use for fastd, but experienced the SQPOLL thread to be very resource consuming mitigating the effect.
io_uring has a feature called FASTPOLL which enables non-blocking sockets to be used. When non-blocking sockets return EINVAL on read/recvmsg in the kernel, io_uring creates a poll for the socket which is what we actually want to keep the stress low. Unfortunately FASTPOLL seems to only work in a busy-waiting thus blocking fashion similar to io_uring's IOPOLL mode.
io_uring currently seems to have problems with mixed read/write attempts somehow leading to a resource busy error.

yanosz, 2020-08-05:
Well, the obvious point is to route non-local traffic using wireguard instead of establishing yet another overlay for batman-adv to underperform. Face it; batman-adv is not meant overlay networks. Most networks have to deal with traffic addressing the internet anyway. Wireguard is built for this, VPN provider offer this service and its about gluon to create such a setup.

You can have batman-adv using an overlay ontop of an overlay. But ... trying to make this performing is squaring circles.

CodeFetch, 2020-08-07:
Little update on fastd with io_uring: It works with kernel 5.7+ and will increase the throughput by at least 20%. Unfortunately it won't work with an older kernel version and OpenWrt master is on 5.4 at the moment. Thus we need to be patient. 

Here is the patched branch: https://github.com/CodeFetch/fastd/tree/final

AiyionPrime, 2021-05-07:
As its not yet mentioned here:
The current plan is to integrate WireGuard using VXLAN and the newly wirtten wgpeerselector package in #2168.

awlx, 2021-09-12:
> We already have VXLAN on the gluon Routers. @NeoRaider @CodeFetch is that not a way to overcom the missing layer2 support of wireguard?

In case it's not known @FreifunkMUC we did exactly this. And it's now also implemented in Gluon.
https://ffmuc.net/freifunkmuc/2020/12/03/wireguard-firmware/

Broker:
https://github.com/freifunkMUC/wgkex/releases

closed 2021-09-14 by blocktrron