freifunk-gluon/gluon #1060: gluon-web: "maxlength" failures leave user clueless

T-X, 2017-03-05:
There are two regressions from Luci to gluon-web regarding the maxlength datatype restrictions:

A) There is no more live feedback when starting to exceed the maximum allowed length. The only failure feedback happens after hitting the "Save" button. Luci gave feedback immediately with the first character which exceeded the allowed length.

B) The feedback given after unsuccessful submission only shows a red background for the text field. There is no feedback regarding what actually went wrong.

NeoRaider, 2017-03-10:
The first issue should be fixed by da19961188a7a60fde2b00282b0ba9a60d83e57f.

NeoRaider, 2020-05-08:
As the instant feedback on validation errors has been fixed, I don't think there is more to do here. There are a few ways to give additional information to the user:
- For a single field: Use the field description string (4th parameter of `option()`) to tell the user what is expected from them (example: key field in `gluon-web-private-wifi/luasrc/lib/gluon/config-mode/model/admin/privatewifi.lua`)
- For global information and errors: Use the form's `message` field (currently not used by any package) or `errmessage` field (used in `gluon-web-admin/luasrc/lib/gluon/config-mode/model/admin/remote.lua`)

I'm turning this into a documentation issue, as these options are currently undocumented.