freifunk-gluon/gluon #1061: Gluon 2016.2.3 - changes via "uci set wireless.radio0.whatever" ignored

biggesee, 2017-03-07:
Hi,
changes via uci set wireless.radio0.whatever seemd to be ignored...
...changes via "luci" for example TXPOWER were accepted 

See also https://forum.freifunk.net/t/gluon-2016-2-3-released/14377/16

Frank

jplitza, 2017-03-07:
Looking at the [list of changes](https://github.com/freifunk-gluon/gluon/compare/v2016.2.2...v2016.2.3), I can only imagine that the merge of #1001 is responsible for the problem. Nothing else even remotely touches that.

Which devices do you see these problems on?

biggesee, 2017-03-08:
nearly* all "0.9.2~" based devices on this list => https://www.freifunk-moehne.net/knotenliste/biggesee/
are involved

* nearly means that I did not check all of them

biggesee, 2017-03-08:
Gibt's hier "verbale Prügel" wenn ich DEUSCHT schreibe? Oder bin ich hier sicher?

A-Kasper, 2017-03-08:
Am 08.03.2017 um 07:44 schrieb biggesee:
> nearly* all "0.9.2~" based devices on this list =>
> https://www.freifunk-moehne.net/knotenliste/biggesee/
> are involved
> 
>   * nearly means that I did not check all of them
> 

Could you provide a mount table? I've got a table where overlay went
ro..... Maybe it's the same problem....

biggesee, 2017-03-08:
root@Olpe-RH311:~# mount
rootfs on / type rootfs (rw)
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mtdblock3 on /overlay type jffs2 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
root@Olpe-RH311:~#

biggesee, 2017-03-08:
**root@Olpe-RH311:/# cat /proc/mounts**
rootfs / rootfs rw 0 0
/dev/root /rom squashfs ro,relatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,noatime 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,noatime 0 0
tmpfs /tmp tmpfs rw,nosuid,nodev,noatime 0 0
/dev/mtdblock3 /overlay jffs2 rw,noatime 0 0
overlayfs:/overlay / overlay rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work 0 0
tmpfs /dev tmpfs rw,nosuid,relatime,size=512k,mode=755 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,mode=600 0 0
debugfs /sys/kernel/debug debugfs rw,noatime 0 0

biggesee, 2017-03-08:
BTW command "iwinfo" is not working...

root@Olpe-RH311:/# **iwinfo**
-ash: **iwinfo: not found**
root@Olpe-RH311:/# 

root@Olpe-RH311:/# **uci show wireless.radio0**
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/qca953x_wmac'
wireless.radio0.htmode='HT20'
wireless.radio0.channel='1'
wireless.radio0.txpower='19'
wireless.radio0.country='DE'

NeoRaider, 2017-03-08:
iwinfo needs to be included in site.mk to be available.

The web interface does exactly the same thing as your UCI command. Please compare the contents of /etc/config/wireless when changing the values via UCI and using the web interface.

viisauksena, 2017-03-08:
it could be that an memory i/o is the cause,
since openwrt(and so gluon) has this bug that if you had one time filled your flash you get always an error and cant write to it until next upgrade. That means that uci commands will do their stuff only on this boot and lost by next reboot. depending on you logic you also only read real uci values instead of temporary ones

biggesee, 2017-03-09:
**iwinfo is included in site.mk** 

(FW-Images were build with the following site.mk wich based on, 
see also https://github.com/larsbu/sites-ffrl/tree/master/site-biggesee)

GLUON_SITE_PACKAGES := \
	gluon-mesh-batman-adv-15 \
	gluon-alfred \
	gluon-autoupdater \
	gluon-config-mode-core \
	gluon-config-mode-autoupdater \
	gluon-config-mode-hostname \
	gluon-config-mode-geo-location \
	gluon-config-mode-contact-info \
	gluon-config-mode-tunneldigger \
	gluon-ebtables-filter-multicast \
	gluon-ebtables-filter-ra-dhcp \
	gluon-neighbour-info \
	gluon-luci-private-wifi \
	gluon-luci-admin \
	gluon-luci-autoupdater \
	gluon-luci-portconfig \
	gluon-luci-wifi-config \
	gluon-next-node \
	gluon-mesh-vpn-tunneldigger \
	gluon-migrate-vpn \
	gluon-radvd \
	gluon-respondd \
	gluon-status-page \
	gluon-setup-mode \
	gluon-tunneldigger-watchdog \
	iwinfo \
	iptables \
	haveged

DEFAULT_GLUON_RELEASE := 0.9.2-biggesee-$(shell date '+%y%m%d')

# Allow overriding the release number from the command line
GLUON_RELEASE ?= $(DEFAULT_GLUON_RELEASE)

# Languages to include
GLUON_LANGS ?= en de

# region information for regulatory compliance
GLUON_REGION ?= eu

#atk10 settings
GLUON_ATH10K_MESH ?= ibss

rotanid, 2017-03-18:
if iwinfo is in your site.mk but isn't on the flashed system you likely have a build problem.
please try again using 2016.2.4 or master branch, AFTER doing a full "make clean" or using a fresh directory.
if the issue persists, search for errors in your build log.

biggesee, 2017-03-20:
my "firmware bulding PC" is down - going to check it later...

biggesee, 2017-03-23:
aus dem FF-Forum:
-----------8<-------https://forum.freifunk.net/t/gluon-issue/14446--------------------------------
[gRaSmOnStEr]

Hi.
Mir ist noch was aufgefallen.. Es werden bei meinem build (egal ob 2016.2.3 oder 2016.2.4) Änderungen, die ich via SSH ausführe, nicht übernommen. ...z.B Client mesh deaktivieren (mit uci set wireless.client_radio0.disabled=1...UCI commit usw) wird zwar ohne Fehlermeldung ausgeführt.. Aber nicht geändert.

Befehle wie reboot...exit funktionieren einwandfrei... Wifi nicht..da hängt sich das WLAN auf.

Ich hab das ganze in einer VM mit Ubuntu Server 16.04 gebaut...
Hab jetzt noch mal alles platt gemacht und von vorne gestartet...gleiche Problematik.
Kann es sein das das eventuell mit der im Forum diskutierten opgk-Quellen Problematik zusammen hängt? Ich bekommen aber auch keine Fehlermeldung beim bauen..

MFG
Christian
-----------8<-------https://forum.freifunk.net/t/gluon-issue/14446--------------------------------

biggesee, 2017-05-03:
@rotanid no changes after rebuliding FW from scratch (now on 2016.2.5)

rotanid, 2017-05-04:
thanks for the feedback, i'm closing this issue. feel free to reopen/open a new one in case.

closed 2017-05-04 by rotanid