freifunk-gluon/gluon #1063: WBS210 txpower limit

Sunz3r, 2017-03-07:
Hi :)

Can't set txpower above 8dBm.

Gluon-Version: gluon-v2016.2-217-g9a33b4c

`
# uci show wireless.radio0.txpower
wireless.radio0.txpower='13'
# uci show wireless.radio0.country
wireless.radio0.country='00'
# iwinfo
client0   ESSID: "Freifunk"
          Access Point: 3A:2E:0B:4A:61:F0
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 8 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
# cat /sys/kernel/debug/ieee80211/phy0/power
13
# iwinfo client0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
*  8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
# iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)
`

rotanid, 2017-03-07:
have you tested, if this is the same when running LEDE ?

Sunz3r, 2017-03-07:
Its build from the current master

kpanic23, 2017-03-08:
I second that! 
I have tested a WBS210 with two 18dBi Yagi antennas connected. Connection was worse than with a 841N v9...
[image]

rotanid, 2017-03-09:
please also test with upstream LEDE images, if these have the same issue, please report the bug upstream and put the link here.

Sunz3r, 2017-03-09:
I can set txpower up to 18dBm with country=US on gluon-v2016.2-217-g9a33b4c:

```
# uci show | grep txpower
wireless.radio0.txpower='13'
# uci show wireless.radio0.country
wireless.radio0.country='US'
# iwinfo
client0   ESSID: "Freifunk"
          Access Point: 3A:2E:0B:4A:61:F0
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 13 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -95 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

inluck, 2017-03-26:
Shouldn't we be able to push this device farther than 18dBm?

It's marketed as being 27dBm.

RubenKelevra, 2017-04-24:
@inluck the powerlimit for US is shown as 20 dB.

Adorfer, 2017-04-24:
I can confirm the issue for 2016.2.5 release for CPE210: txpower in iwinfo-output always at 8dBm. 
(not clear if it's not adjustable, not change if txpower set in /etc/config/wireless to 5,10,18 or 20 or unset. Or if it's just an issue with the output of iwinfo.)

Sunz3r, 2017-06-09:
It seems to be fixed in current master (for WBS210):

```
# iw reg set DE
# iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 14), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)
# iw phy phy0 set txpower fixed 2000
# iwinfo
client0   ESSID: "Freifunk"
          Access Point: 3A:2E:0B:4A:61:F0
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -95 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
# cat /sys/kernel/debug/ieee80211/phy0/power
20

```

A-Kasper, 2017-06-09:
Does the TX-Power of iwinfo include the antenna gain?

Sunz3r, 2017-06-09:
I think not ... i tested it with different antennas and hardware.

however i have bad mesh-links, shorter range and worse RX-MCS-Values with WBS210 ... so i replace it with WR841 till it is fixed.
higher txpower doesnt change this issue

rotanid, 2018-11-25:
this upstream issue will be tracked in Gluon over there: #1156
closing.

closed 2018-11-25 by rotanid